Ultimaker/Cura #4127: feature request: per-object profiles

tmbdev, 2018-07-24:
**Application Version**
3.3.1

**Platform**
Linux

**Printer**
independent of printer

**Steps to Reproduce**
1. load/replicate multiple objects, select "one at a time printing"
2. select an object
3. change profile

**Actual Results**
profile changes for all objects

**Expected results**
profile only changes for selected object (maybe with dialog/menu entry)

**Additional Information**
When printing multiple objects one at a time, it can be very useful to use different print profiles for printing each object, either to optimize print settings (when printing multiple benchmark objects) or for different parts requirements (large structural, small functional). I didn't see any obvious ways of accomplishing this in the current UI. It would be nice if there was some simple support for this.

ChrisTerBeke, 2018-07-24:
You can already adjust setting per object by selecting the object and clicking on 'Per model settings' in the toolbar on the left.

Switching complete profiles would not really be possible as objects might then have different layer heights, flow rates or other things that are hard to switch mid-print (while keeping proper printed part quality).

tmbdev, 2018-07-24:
It should be easy to switch profiles mid print for objects that are printed one at a time. They are essentially like separate prints on the same build plate

ChrisTerBeke, 2018-07-24:
Cura is not designed to do this, so no, it's not easy. I doubt this functionality will be worked on at Ultimaker anytime soon.

cc @Ultimaker/cura

BagelOrb, 2018-07-24:
Slicing it is easy, all the machinery is there. But the frontend has quite some complicated code to workout what settings should follow from a profile. Making the profiles per object makes it all so much more difficult. The settings one should use should then depend on the object, the extruder, the material and the machine.

Ghostkeeper, 2018-09-03:
The problem with this feature (that we stumbled on before) is that a profile created for extruders might change global settings that we can't set per mesh, such as the build plate temperature. The only real fix is to add a whole interface and architecture for creating and managing per-object profiles separate from ordinary profiles. That is just a bit complex for little gain.

fieldOfView, 2018-09-03:
If it is implemented in a way that can also work for The postprocessing plugin (which also has a stack of settings), it would be very usefull though.

tmbdev, 2018-09-09:
There are several different ways in which people might want to have "per object settings". For objects that are printed all at once, I might want to vary some slicing settings between them: speed, extruder, infill, printing temperature, etc. For objects that are printed one at a time, I might want to vary many more settings, like print bed temperature, layer height, etc.

The per-object settings for all-at-once printing and the per-object settings for one-at-a-time printing are both different in nature and implemented differently. In fact, while per-object settings for all-at-once printing may involve substantial changes to the slicer, the per-object settings for one-a-time printing are really more like a concatenation of otherwise independently sliced printjobs. The only reaons to expose this through the Cura interface is that as a user, I still need to place the individual objects on the same build plate.

So, to put this differently, what I'm really suggesting is something more like a "multipart job" for one-at-a-time printing. Each part of the multipart job would get sliced with its own settings. There would be no impact on the slicer at all.

Ghostkeeper, 2020-12-01:
Duplicate of https://github.com/Ultimaker/Cura/issues/3192.

closed 2020-12-01 by Ghostkeeper