Ultimaker/Cura #4129: Cura Projects not showing on the Recent Files list

SuperconductingCarbon, 2018-07-25:
**Application Version**

3.4.4
**Platform**

Linux a6 4.4.0-130-generic #156-Ubuntu SMP Thu Jun 14 08:53:28 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux
**Printer**

Creality CR-10S
**Steps to Reproduce**

Save as a  Project.  It does not show in the "Recent Files" list. 
**Actual Results**

Only .stl files show in the "Recent Files" list.
**Expected results**

Expect to see the Project files and well as STL's in the Recent files list.
**Additional Information**

<3 Cura!

alekseisasin, 2018-07-27:
@SuperconductingCarbon , Thank you for your feedback. 
We will implement this improvement. 
DEVS: CURA-5601

Ghostkeeper, 2018-09-14:
This should be implemented for the next release.

closed 2018-09-14 by Ghostkeeper