Ultimaker/Cura #4131: Z Seam user specified position different on 3.4.1 from 3.3.1

amagr0, 2018-07-25:
**Application Version** - CURA 3.4.1
**Platform** - Win 10 Pro, 64 bit
**Display Driver** - NVIDIA GeForce 840M

**Steps to Reproduce** - Generate a gcode with the following settings:

- Z Seam Aligment: User Specified

- Z Seam X: 0

- Z Seam Y: -100

**Actual Results** - The Z Seam is not on the same position of the same model generated on Cura 3.3.1

**Expected results** - The Z Seam position on the indicated coordinates, like the Cura 3.3.1

[image]

[Cura_Projects_341_331.zip](https://github.com/Ultimaker/Cura/files/2227597/Cura_Projects_341_331.zip)

amagr0, 2018-07-25:
And also on the 3.4.1 the Z Seam is not vertical aligned.

[image]

smartavionics, 2018-07-25:
Hi, I just sliced your 3.4.1 project using my master branch Cura (with my mods) and the z-seam is where it should be and it looks reasonably aligned.

[image]

I also sliced it with a 3.4 based release (without my mods) and the result looks pretty much the same.

Sorry, I cannot reproduce the issue from your project file.

diegopradogesto, 2018-07-26:
Loading the file Piramide_50_Cura331.curaproject.3mf, this is the result in 3.3.1:
[image]

And this is what I see in 3.4.1:
[image]

Scrolling all the layers, they seem to be the same, so I don't see any issue here.

closed 2018-08-22 by amagr0