Ultimaker/Cura #4134: How to create curapackage files

thopiekar, 2018-07-25:
Hey guys,

do you have any info around the curapackage file?
Looked for docs what is new in this fileformat, but could not find anything.
Also in Uranium there is still the old script for building curaplugin/umplugin files.

An example file would be great, too.

Thanks!

ChrisTerBeke, 2018-07-26:
There's a Wiki page about it.

BUT..... we'll launch a managed service for this very shortly to build and distribute your packages from your source files. I can give you beta access if you want to try it out soon.

fieldOfView, 2018-07-26:
Gimme, gimme!

ChrisTerBeke, 2018-07-26:
Expect an invite in your inbox in about 2-ish weeks :)

thopiekar, 2018-07-26:
Awesome 🤘

fieldOfView, 2018-07-29:
It is not too hard to reverse engineer a current package...

thopiekar, 2018-07-29:
Sure, but if people say that there is a document describing it already, why should I reverse engineer it?

ChrisTerBeke, 2018-07-29:
https://github.com/Ultimaker/Cura/wiki/Creating-Packages

ChrisTerBeke, 2018-07-29:
Closing this as ticket (we can still continue to discuss about packages).

closed 2018-07-29 by ChrisTerBeke

thopiekar, 2018-07-29:
Thanks!

There is the first question already. Does the directories name inside ```plugins``` correspond to the plugin's id?
So ```"MyPlugin" == "ExamplePackage"``` or is there no convention how the directory inside ```plugins``` is called?

[image]

ChrisTerBeke, 2018-07-29:
Yes, it does :)

Our upcoming tooling will check the directory names and required filenames in there as well (and tell you what's wrong if one of them is not there).

thopiekar, 2018-07-30:
Ok, and what does the 2.7.0+ mean? Is there a check per Cura version? Is it possible to roll new releases for older Cura versions out? Eg. only with bugfixes?

ChrisTerBeke, 2018-07-30:
You can indicate which Cura versions a release of your package is compatible with. You can simply make a new release of your package with bug fixes and not click the toggle for a newer Cura version. It's very App Store / Google Play like :)