Ultimaker/Cura #4140: start and end of print blob issue and feature request

kornshadow097, 2018-07-26:
**Application Version**
3.4.1

**Platform**
windows 10

**Printer**
tevo tarantula

**Steps to Reproduce**
print the shape. layer preview doesn't show this
result.
1If I'm printing a continuous shape like a cube with round edges I get a start 
and end stop blob every single layer from top to bottom on both the starting 
spot and end spot has random layer scarring with a mini blob on the outer wall.

2)Another issue, sometimes when i have multiple instances of cura open my machine 
settings will back to default if any of those settings were changed in another
profile rather than syncing between all softwares...

also if you try to update a setting across them, you have to activate each one
and this takes forever... why do we have to activate them to update settings?
I still say adding nozzle size to a menu next to quality and Layer height is a 
better idea than having to set it under a machine ... which can at anytime change
its nozzle and make the profile obsolete and make errors (experienced this)

**Actual Results**
 
1)i  get random blobs.
 2)I lose settings and updated changes to machine or profiles.

**Expected results**

**Additional Information**

[TT_single_wall_box.curaproject.zip](https://github.com/Ultimaker/Cura/files/2232975/TT_single_wall_box.curaproject.zip)

kornshadow097, 2018-07-26:
1) i started using linear advance and was able to move that irregularity into the center of a wall vs on the outside.. it seems jerk or k factor are able to move it. retraction may have helped too along the way

ChrisTerBeke, 2018-07-30:
Closing in favour of separated issues.

closed 2018-07-30 by ChrisTerBeke