Ultimaker/Cura #4141: Incorrect MIME type for STL files

unlimitedbacon, 2018-07-26:
**Application Version**

3.4.1

**Platform**

Arch Linux
(I believe this also affects Ubuntu 18.04 and Debian 10, since they both use shared-mime-info 1.9)

**Printer**

N/A

**Steps to Reproduce**

View or open STL files in the system's file manager after installing Cura.

**Actual Results**

After installing Cura STL files no longer have an icon and are no longer opened by applications besides Cura.

**Expected results**

STL files should retain the icon from the system theme and the "Open With..." menu should allow you to open the file with any supporting application, including Cura.

**Additional Information**

I recently noticed that my file manager was no longer recognizing STL files properly. The correct icon was no longer being displayed and they were not being opened with the correct applications. Upon further investigation I discovered that Cura is the culprit.

Cura is designating STL files with an incorrect MIME type. Cura installs a shared-mime-info package (`cura.xml`) which marks STL files with the type `application/sla`. This overrides the default type for STL files, which is `model/stl`.

`application/sla` is an older unofficial type which was probably used since previously there was no official assignment from the IANA. Since March, the IANA has settled on the type `model/stl` as the official type for STL files.
https://www.iana.org/assignments/media-types/model/stl

`model/stl` has been included in the shared-mime-info core database since version 1.10. Previously, shared-mime-info also had `model/x.stl-binary` and `model/x.stl-ascii`.
https://cgit.freedesktop.org/xdg/shared-mime-info/commit/?id=99ed700e1e2c9118e300c7e257618f4ecb8804ec

BagelOrb, 2018-07-26:
Interesting stuff. Should be a one liner fix, right @ghostkeeper ?

ianpaschal, 2018-07-31:
Maybe. We had a similar issue on Windows which was a real pain to fix but I guess in theory that's a lot easier on Linux.

Ghostkeeper, 2018-09-21:
Yes, clearly documented issue and why it must be improved. Thanks a lot for notifying us of this new MIME type, @unlimitedbacon! It'll be fixed in version 3.6.

commit 30ef7243 (2018-09-21): Update STL MIME type

Since march, there is now an official MIME type for STL files. See https://www.iana.org/assignments/media-types/model/stl for the new MIME type specification.

Fixes #4141.
closed 2018-09-21 by Ghostkeeper