Ultimaker/Cura #4143: Add number of layers, or model height, to gcode variables

Jean-ClaudeD, 2018-07-26:
versions: All
Platform: assume all, but windows specifically
Printer: NA

The existing **start.gcode** allows options like  {day} {date} {time} as a form of variable replacements.

I'm suggesting one or two new options be added:
{LayerCount} and/or {TotalLayerHeight}

Why?
The existing default **start.gcode** file includes an entry:
**M117 Printing..**

It would be nice to add a **M117 Printing {LayerCount} layers..**

The value for the **{LayerCount}** is already being computed as show in this gcode sample:

> G21        ;metric values
> G90        ;absolute positioning
> M82        ;set extruder to absolute mode
> M107       ;start with the fan off
> G28 X0 Y0  ;move X/Y to min endstops
> G28 Z10     ;move Z to min endstops
> G1 Z15.0 F10500 ;move the platform down 15mm
> G1 E10   ; start the flow of plastic
> G92 E0                  ;zero the extruded length
> G1 F200 E3              ;extrude 3mm of feed stock
> G28 X0 Y0  
> G92 E0                  ;zero the extruded length again
> G1 F10500
> ;Put printing message on LCD screen
> M117 Printing..
> 
> ;Layer count: 558
> 

My request would allow someone to add a field if desired to the start gcode to say 
> M117 Printing 475 layers..

the added gcode being something like:
**;Put printing message on LCD screen
M117 Printing {LayerCount} layers..**

So, as the model is printing, you can get an idea of how far along it is, from the status screen.  An even better option might be:
> M117 Printing 47.5 mm total...

the added gcode being something like:
**;Put printing message on LCD screen
M117 Printing {TotalLayerHeight} mm total..**

ChrisTerBeke, 2018-07-26:
I don't think this will fit on most LCD displays. They have a character limit of 16, so the default "M117 Printing.." just fits in that... (e.g. https://www.adafruit.com/product/716).

ChrisTerBeke, 2018-07-26:
Ofc you can always override the default start gcode with your own if you want this behaviour. I'm not sure though which variables are available in the start/end gcode snippets.

Jean-ClaudeD, 2018-07-26:
My point is I don't think there are variables that hold this information.  I'm using a Anet A8 clone, and I've set this to 20 characters without issue.
In the print I am doing now, it tells me I am at a Z height of 34.1, but I have to look back at the computer to find out the model being printed is 74 mm tall, and that I am therefore 45% completed.  If I could display the number of rows, or the overall height, I could mentally compute it without having to go back to Cura to view what is printing by layers, to the height then multiply it by the layer height.

ianpaschal, 2018-08-01:
There obviously is a variable for this, which CuraEngine inserts in the `;Layer count: 558` line. The problem is actually adding more settings to let the user specify some additional text which is sent to Cura Engine, as well as then parsing that text on the engine side, extracting variables and swapping in the correct values, in order to deliver the code back.

Considering the amount of work required, I think it's unlikely that we'll do it, but we will discuss it.

fieldOfView, 2018-08-01:
This can be implemented fairly easily with a post-processing script.

Jean-ClaudeD, 2018-08-01:
> variables in gcode slicer (#4143)
> <author@noreply.github.com>

Really? I was not aware of that ability when using Cura with the built in slicer and USB printing engines. Can you point me to any resources or documentation?

fieldOfView, 2018-08-02:
The variable replacement happens before slicing, so the number of layers is not available yet. Also the variable replacement only works on the start and end gcode snippets.

Jean-ClaudeD, 2018-08-02:
In the default install, under the gcode start, it shows:    ;Filament used: {filament_amount}m {filament_weight}g    ;Filament cost: {filament_cost}If these values are replaced BEFORE the model is sliced, how do you know what these values would be, since it isn't until it is sliced that you can even begin to know how much filament is needed.

ianpaschal, 2018-08-02:
Yeah we can add it. I forgot that you could put variables in the machine start and end gcode slots.

Ghostkeeper, 2018-09-14:
> Someone suggested this as a post processing script.  Do you have any information on how I would start on such a task?

I would implement it using the following algorithm (this is pseudocode a.k.a. untested Python):

    for layer in data:
        for line in layer:
            if line.startswith(";LAYER_COUNT:"):
                layer_count = line.split()[1]
                break
        else:
            continue
        break
    data[0] = data[0].replace("{layer_count}", str(layer_count))

You have to look at other post-processing scripts to see the framework on which they are built. The search-and-replace script is always an easy one to start with.