Ultimaker/Cura #4146: Package install

FynDestiny, 2018-07-27:
**Application Version**

3.4.1

**Platform**

Windows 10

**Printer**

not applicable

**Steps to Reproduce**

Toolbox/browse packages
select solidworks
install

**Actual Results**

Goes through the install procedure etc. Doesn't appear to show any problems.
Once Cura is restarted, the package doesn't show.
Back in the package browser, Solidworks shows as "install" not "installed"
And there is an extra button in Solidworks to "Export to Cura". Click on it and Solidworks starts Cura, but the file conversion the package is supposed to do, doesn't happen.

**Expected results**

Expected to see the Solidworks package option in the Extensions menu.
Back in the package browser, Solidworks shows as "installed" not "install"

**Additional Information**

This package installed into Cura in the previous version, but did not show anything in Soldidworks itself.

This time, although it does not appear to install into Cura, there is now an extra button in Solidworks to "Export to Cura"

The screenshots were taken in order of procedure to install package and try to use.

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

FynDestiny, 2018-08-09:
Hi
Sorry for late response.
It appears to have installed properly now.
I just tried to upload the log file, but it's too big

diegopradogesto, 2018-08-10:
OK, no worries, let me know if it still happens.

closed 2018-08-10 by diegopradogesto