Ultimaker/Cura #4147: Print "One at a Time" option

FynDestiny, 2018-07-27:
**Application Version**

4.3.1

**Platform**

Widows 10

**Printer**

Creality CR-10S

**Steps to Reproduce**

Added 5 items.
Select "Extra Fine" print quality.
Selected "One at a Time" option.
Click Prepare.
Select Layer View.

**Actual Results**

Prepares Gcode with no apparent issues with 1472 layers (screenshot 1)
Scanning through the layers, the last item completes at layer at 1156 (screenshot 2)
Layer 1157 is shown in screenshot 3, and you can see strange lines from the bottom left to one of the items

**Expected results**

Expected result would be the Gcode ending at layer 1156, not 1472

**Additional Information**

[CCR10_Base.curaproject.3mf.zip](https://github.com/Ultimaker/Cura/files/2235749/CCR10_Base.curaproject.3mf.zip)
[image]
[image]
[image]

BagelOrb, 2018-08-02:
This is most probably just a visualization issue.
If I scroll beyond 1152 then suddenly those lines disappear again.

Perhaps your graphics card just gets too many lines or something.

The lines don't appear to be in the gcode either:
[image]

BagelOrb, 2018-08-02:
I'm sorry, you're right. The engine probably communicates the wrong number of layers to the frontend.

rburema, 2018-09-28:
Hi @FynDestiny, I've loaded/sliced your project in the current master and 3.5-BETA, and in both cases it stops at layer 1155. Might be this was related to another issue that we fixed in the meantime. Could you perhaps try and reproduce this in the 3.5-BETA? cheers, Remco.

Ghostkeeper, 2020-06-30:
This still happens in 4.7. I know from experience.

GregValiant, 2024-11-15:
Is this still an issue with current Cura versions (5.8.0 and up).  Can this be closed?

closed 2024-12-09 by github-actions[bot]