Ultimaker/Cura #4151: Unclearness about file readers

thopiekar, 2018-07-27:
I see different implementations for many readers. I don't know whether this is intended or not.

1. Mimetype defined for GCode reader, but GCodeGZ reader does not define any. However, the defined mimetype inside the GCode reader defines the .gcode.gz, too. But the GCode reader itself does not handle but GcodeGZ and that one depends on the GCode reader.
In short: That does not make sense to me. The definition of the gz compressed file should go into the GCodeGZ reader, in my point of view.

* https://github.com/Ultimaker/Cura/blob/master/plugins/GCodeGzReader/GCodeGzReader.py
* https://github.com/Ultimaker/Cura/blob/master/plugins/GCodeReader/GCodeReader.py

2. Mimetypes are either defined on the first level or inside the reader's class' __init__ function. Examples are:
* https://github.com/Ultimaker/Cura/blob/master/plugins/3MFReader/ThreeMFReader.py
* https://github.com/Ultimaker/Cura/blob/master/plugins/GCodeReader/GCodeReader.py
and the X3D Reader does not define any MimeType.
* https://github.com/Ultimaker/Cura/blob/master/plugins/X3DReader/X3DReader.py
Due to what I observed on the CAD plugins, I guess that the print project will be called "unknown" on Windows (!). Doesn't happen on Linux.
Personally, I would implement the Mime registration call like in ```GCodeReader.py```, because then the type will be only registered once when the plugin gets loaded.

Anyhow, many thanks for taking time into answering the questions already! 👍

ChrisTerBeke, 2018-07-30:
I think the issue of 1) is that we can only define a mimetype in 1 location, so doing it in gcodegzreader as well would not work as there's already a mime entry for gcode.

Ghostkeeper, 2018-09-27:
Second one is also fixed now. The MIME type declaration should not be made statically in plug-ins, since the static code is executed upon importing the .py file and that file is always imported, regardless of whether the plug-in has an outdated API version or when it is disabled. I've moved these declarations into the constructor of the plug-in objects.

One exception was the XmlMaterialProfile plug-in, which defines it in the register function. I couldn't get around that one because the plug-in objects are created thousands of time, once for every plug-in. Don't want to add the MIME type thousands of times!

commit 88ba2ac3 (2018-09-27): Define gcode.gz extension in GCodeGzReader

So if you were to disable the GCodeGzReader plug-in, you will now no longer see the extension in the files you can read.

Fixes #4151.
commit 0ce9bf61 (2018-09-27): Move MIME type declarations into constructors of readers

So that if you disable the plug-in, the MIME type declaration is also not added.

Fixes #4151.
closed 2018-09-27 by Ghostkeeper