Ultimaker/Cura #4154: CATIA STL Scaling issue

Aeroengineer1, 2018-07-29:
When parts are exported from CATIA as an STL and then brought into CURA, the default scaling is very off.  I can say that it seems to exist in V3.X of the program, but I am not sure how far it goes back.  

The models come in and scale to 10,000x, but to get them to the correct scale, they need to be scaled by 2540x.  This is an odd scaling factor.  It is related to metric to inch conversion with the 2.54 factor that is in there, but not sure where the 1000x comes from.  It is essentially bringing in the model in micrometers.

[image]

You can see the automatically scaled model, next to the properly scaled models.

ChrisTerBeke, 2018-07-29:
The size is indeed something you have to define when exporting. Or you can use Cura's scaling tool to correct the scale.

closed 2018-07-29 by ChrisTerBeke

Ghostkeeper, 2018-09-17:
The x1000 is added when the model is so small that Cura is taking the assumption that the units were exported in metres. It converts it to millimetres then.