Ultimaker/Cura #4155: Slice Lines in Air

Dj-EKI, 2018-07-29:
**Application Version**
3.4.1

**Platform**
Windows 10

**Printer**
Anet A8
Ultimaker 3

**Steps to Reproduce**
It produces lines in the air.
[image]

From layer 28 to layer 29, the lines are wrong. No support.
[image]

**Additional Information**
I had already created a report for you, but this time I can upload a project file.

[CFDMP_RadioLenovoLinks.curaproject.3mf.zip](https://github.com/Ultimaker/Cura/files/2238603/CFDMP_RadioLenovoLinks.curaproject.3mf.zip)

I have now made a few advanced settings to try to fix this problem, but I expect from a program that the default will work fine too.

smartavionics, 2018-07-29:
You need to enable the support interface. Here's some example settings:

[image]

Dj-EKI, 2018-07-29:
Thanks, a problem is fixed. The airlines are unfortunately still to fix.

[image]

smartavionics, 2018-07-29:
Sorry, I can't see where that part is on the model, please post an image that isn't so zoomed in.

Dj-EKI, 2018-07-29:
It produces lines in the air. (Layer 71, Object Rotate 90°.)

smartavionics, 2018-07-29:
Please post another image not so zoomed in because I cannot recognize that part of the model from the image above.

Dj-EKI, 2018-07-29:
[image]

Dj-EKI, 2018-07-29:
If I use tree, then I could fix this error. But it does not do too much support and some of it is not very good.

smartavionics, 2018-07-29:
[image]

Dj-EKI, 2018-07-29:
Thanks, Support Density fixes my mistake. This issues can be closed.

smartavionics, 2018-07-29:
Great, glad it's fixed. I can't close this issue but you can do that. Cheers!

Dj-EKI, 2018-07-29:
Unfortunately, I still have a problem. Why is my support not filled.
[image]

smartavionics, 2018-07-29:
You have set support density to 2%!

closed 2018-07-29 by ChrisTerBeke