Ultimaker/Cura #4156: [3.4.1] Save Project - There are no file types available to write with!

zephyweb, 2018-07-29:
**Application Version**
3.4.1

**Platform**
Arch Linux x86_64
Cura package: [https://www.archlinux.org/packages/community/x86_64/cura/](url)

**Steps to Reproduce**
Click File in the top menu
Click Save Project...

**Actual Results**
Error
There are no file types available to write with!

**Expected results**
A file save dialog to choose the name and location of the project file.

**Additional Information**
This is exactly the same issue as: https://github.com/Ultimaker/Cura/issues/3409, but with libsavitar-3.4.1-1 already installed ([https://www.archlinux.org/packages/community/x86_64/libsavitar/](url))

ChrisTerBeke, 2018-07-29:
Ultimaker does not maintain that Arch package, but given the error message, it seems that libSavitar cannot be found (which is needed to read/write 3MF files).

ianpaschal, 2018-08-03:
Probably they did not install it correctly in that case. Regardless, we can really only offer support on binaries we built. Even if we were able to debug it, you'd have to contact the maintainer to fix it.

Sorry, you'll need to contact the maintainer of that package and if they have specific questions about their build environment maybe we can help, but there's not really anything we can do about this.

closed 2018-08-03 by ianpaschal

zephyweb, 2018-08-08:
Thanks for your help ; for information this bug was reported here : https://bugs.archlinux.org/task/59490?project=5&string=cura