Ultimaker/Cura #4158: Cura TAGs

mtmello2003, 2018-07-29:
Hi,

Where can I find the complete list of cura tags (variable) that we can use in gcode scripts? 

like this for example:
{material_print_temperature_layer_0} 

Thanks

BagelOrb, 2018-07-30:
Cura/resources/definitions/fdmprinter.def.json 
There are also maybe two settings defined in fdmextruder.def.json

closed 2018-07-30 by BagelOrb

ChrisTerBeke, 2018-07-30:
This question keeps coming back though, maybe we should add a section to the Wiki.

fieldOfView, 2018-07-30:
Since this keeps coming back every so often, I made a quick and dirty html page that parses fdmprinter.def.json and fdmextruder.def.json and spits out a list:

http://files.fieldofview.com/cura/Replacement_Patterns.html

https://gist.github.com/fieldOfView/03636e14a1889d73b331cab2c3e06040#file-replacement_patterns-html

Ghostkeeper, 2018-09-18:
Apart from all the settings, we also expose the `{date}` and `{time}` and also some old names of settings. See https://github.com/Ultimaker/Cura/blob/6a053a9f46c56a94cfaf611cb55376b4d0ad09d1/plugins/CuraEngineBackend/StartSliceJob.py#L275 for the implementation.

fieldOfView, 2018-09-18:
I had already added those to the top of the document linked above.