Ultimaker/Cura #4161: cura keeps crashing

michael12515, 2018-07-30:
**Application Version**
3.4.1

**Platform**
 windows 10

**Printer**
tevo tarantula

**Steps to Reproduce**
open cura, click the button to load stl files , click the pop up window and then cura said that its not responding and closes.

**Actual Results**
it crashes

**Expected results**
it should let me find the stl file and load it

**Additional Information**

ChrisTerBeke, 2018-07-30:
Please share the log files as indicated by the template and README.

michael12515, 2018-07-30:
alright thanks

michael12515, 2018-07-30:
i sent it through we transfer

ChrisTerBeke, 2018-07-30:
To which email address??

It's easier to track if you log in to GitHub and upload the file in this issue thread.

michael12515, 2018-07-30:
[log.txt](https://github.com/Ultimaker/Cura/files/2242869/log.txt)

Timlindz, 2018-08-01:
Same problem to me cura 3.4.1 crashes anytime it tries to open any windows that isn't a normal cura dialogue box such as if you go to settings then click manage profiles and a cura box opens but if you then click import it opens a normal windows box to select a profile to import and then seconds later crashesit happens every time it opens a normal windows box instead of a cura box. even when you click save file and it opens a windows box folder to save the gcode file to and then crashes.  
[windows info.txt](https://github.com/Ultimaker/Cura/files/2247766/windows.info.txt)

Ghostkeeper, 2018-09-21:
This is a duplicate of #1537.

closed 2018-09-21 by Ghostkeeper