Ultimaker/Cura #4163: Outermost top layer ignoring material flow % setting

hanswwilson, 2018-07-31:
**Application Version**
3.4.1

**Platform**
Windows 10 Pro

**Printer**
Prusa i3

**Steps to Reproduce**
1. Setup printer as though it had two extruders
2. Set Top/Bottom line width to something wider than normal (tested with values between 0.6mm and 1.0mm on a printer with a 0.4mm nozzle)
3. Set Top/Bottom Extruder to the second extruder
4. Set Top Surface Skin Layers to 0 and just in case also set Top Surface Skin Extruder to the second extruder
5. Set a number of top layers (tested with 3, 4 and 5)
6. Set bottom layers to 0
7. Set the second extruders material flow % to a low value to cause intentional under-extrusion (tested with values between 80% down to 40%)
8. Allow print to slice. Object should show as perimeters and infill being printed with the first extruder and the top layers printed with the second extruder
9. Generate gcode file
10. In order to print this on a machine with one extruder manually edit the gcode file removing all instances where it swaps extruders (in my case I replaced all instances of the command T1 with T0)
11. Print file

**Actual Results**
The initial top layers print with the expected under-extrusion due to the low material flow %, however the final outermost top layer does not print with the expected under-extrusion due to the low material flow %

**Expected results**
The final outermost top layer following the behavior of the initial top layers should print with the expected under-extrusion due to the low material flow % specified.

**Additional Information**
Photo of prints stopped halfway through printing the final outermost top layer demonstrating the lack of under-extrusion expected from the low material flow % specified. The initial surface layers print the grid like structure cause by the wide line width and purposeful under-extrusion while the final top layer fills in solid.
[image]

The goal with this surface structure is to use the wide line widths and under-extrusion to generate surfaces that are sieve or grid like in nature and will allow air to pass through. These parts can then be used as molds in the thermoforming process drawing the material being formed tightly against the surface while a vacuum is drawn through the printed part. This would be highly beneficial to people using 3d printed forms for thermoforming.

While a fix to this would be great a better solution would be that addition of some kind of setting for a mesh or sieve infill that would allow you to specify a gap width between the top layer lines in the Lines pattern. This would allow you to directly generate surfaces like this with out needing to resort to line width/under-extrusion/multiple extruder tricks.

BagelOrb, 2018-08-02:
Please always include a project file.

Did you look at the layer view to verify that it is a software problem?
It could be a hardware problem as well.

BagelOrb, 2018-08-02:
[image]
This shows that Cura doesn't do anything wrong.

I think your printer is loosing pressure, I suggest increasing the Top/Bottom Line Width to 1.0mm and have the flow at about 40%, so that the actual lines being printed are still equal to the nozzle size.

hanswwilson, 2018-08-03:
I dont fully understand what your image is showing, as its the solid surface not the infill that I am talking about. One of the examples in my image was done at 1mm with 40% infill and the .8mm done at 60% would have resulted in a line that was wider than the nozzle, and those still had the same issue.

All that aside, in order to make a project file I completely rebuilt my printer profile and all the print profiles from the ground up and of course having done that, I am no longer experiencing the issue as I was before. Based on that there must have been some setting somewhere that was causing what was happening to occur. It is still an absolute struggle to get the mesh consistent across the entire surface but the outermost top layer filling in solid as depicted in the photos in my original post is not happening like it was before, having remade the print profiles from scratch. 

I would still like to have this as a feature request to include some option to set the spacing between the lines in the top/bottom pattern. Having to resort to all kinds of workaround and tricks when a single setting could control the spacing is frustrating. Whether this should be closed and a new issue opened as a feature request of if this can just be changed to be the feature request I don't know.

[image]

BagelOrb, 2018-08-03:
It's better to make a separate ticket for the feature request.

It's a very simple feature, but your reasons for wanting it are pretty unique, so I'm not sure of the priority of the this feature compared to other issues.

You could also get the same effect using gradual infill or with infill meshes (per model settings).

hanswwilson, 2018-08-03:
Not sure how gradual infill would work as only my test object is a nice flat square. Actual things being vacuformed are a lot more complex in shape so the top/bottom settings are still important. The per model settings stuff seems even more complex to setup than the workarounds I am doing now, especially when the shapes of parts get more complex.

Thermoforming or vacuforming is a common manufacturing method so I would not consider it a particularly niche use case. Using 3d printers to generate the forms seems like a great thing to support. Most forms have to be cnc milled or sculpted. Being able to make the forms with airflow already built into the parts is a significant improvement to the process.

BagelOrb, 2018-08-03:
Well I would consider you a pioneer of 3D printing for vacuforming. :)

Gradual infill is made to work with any geometry; you should give that another try. Tip: make the top thickness 0% and the infill 50% (at the top most region).

With per-model settings you can easily separate the top most part by loading your model twice and having the modifier mesh sunk down a bit.

Please give these options a shot; they come a lot closer to what you want than what you are currently trying.

hanswwilson, 2018-08-03:
Gave gradual infill a look and it seems like it also would sort of work. It is a little easier to have it generate what I am looking for as I don't have to edit gcode files or rely on calculating widths and under extrusion as the infill line distance can be precisely set. The thickness of the surface is settable using the infill step height as well which is nice. Tight grid patterns like I am getting using the width/under-extrusion stuff need a 90-95% infill density though, and the way the infill steps work means the parts end up much denser than they really need to be. There are two other things I ran into though that are causing me problems.

First issue is when I have gradual infill enabled and no surface layer I get all these little chunks of infill around the edges of the part. They really do a number on the print time and on the printer as its trying to deal with all the short movements. I have not been able to find any way to not have those when using the gradual infill.
[image]

Second issue I have encountered with gradual infill is that it only fills right to the edge of any surface feature above a flat layer, so the layer adhesion for the perimeters above can be severely affected as not all the infill is supported nor are the perimeters above it. Is there anything like the Skin Expand Distance setting for the infill that would push the more solid infill pattern deeper into perimiters?
[image]

This is what it looks like using the Skin Expand Distance setting to ensure that any perimeters above a change in surface are adequately supported. The surface goes quite deep into the perimeters so the chance it has 
[image]

BagelOrb, 2018-08-06:
If those little infill bits are really caused by there being no top layers than I wouldn't know a way around that bug.
However, if that's not the case it might be caused by Fill Gaps Between Walls.
I think there's infill there to support a chamfer at the top of your model, though.

For such an expansion on infill, use Infil Overlap

Ghostkeeper, 2018-10-09:
We've discussed this issue with the team and deferred it. We don't consider it more pressing than other things we have to do, because the extruder-per-feature is not used very often.

It was already known to us previously from our own tests as well. The flow is taken from the object's extruder rather than from the extruder that the feature is printed with, because the flow is not considered to be specific to a particular feature such as skin or infill. For all other settings this works out fine because they are not adjustable per line or because we have separate settings for it per feature and we use those instead.

closed 2023-06-24 by github-actions[bot]