Ultimaker/Cura #4165: Brim on inside goes outside the buildplate to infinity [3.4.1]

printingotb, 2018-07-31:
**Application Version**
3.4.1 (also 3.4.0)

**Platform**
win7 64bit

**Printer**
Custom FDM printer
(Wanhao Di3)

**Steps to Reproduce**
Load my project file and slice with the "brim only on outside" checkbox unchecked.
Each time I slice this I get different magnitudes of the issue. Rechecking and unchecking the box gives a new pattern and print-time when sliced again. 
Slicing with it checked works as intended.

**Actual Results**
The brim goes out of the build area and adds a lot to the print time an material used, See attached pictures

**Additional Information**
I have checked the stl with Meshmixer for errors.
I attached pictures and my project-file
[image]
[image]

[Brim_inside_issue.curaproject.zip](https://github.com/Ultimaker/Cura/files/2246936/Brim_inside_issue.curaproject.zip)

Let me know if I need to add more info.

BagelOrb, 2018-08-02:
I cannot reproduce this on current master, but I cannot load the project file in 3.4.1.

That being said, such a bug could be specific to your computer hardware setup.

Can you attach the gcode?
It could be the problem is in the visualization only.
Here's a nice online gcode viewer you don't need any installation for: http://www.chilipeppr.com/tinyg

printingotb, 2018-08-02:
Thanks for the reply! 
I started the print and it did actually try to travel outside the buildplate. It was travelling and not extruding. Probably on it's way to start extruding at the other end of the line.
It was also printing a "cross" on top of the already printed brim (at same layer).
Here is the G-code I just ran:
[Buggy.zip](https://github.com/Ultimaker/Cura/files/2253574/Buggy.zip)
And here is the stl (I made a new similar to the original since customer project, both have the issue):
[BrimIssue.zip](https://github.com/Ultimaker/Cura/files/2253607/BrimIssue.zip)
[image]

BagelOrb, 2018-08-02:
There's definitely something wrong in master:
[image]

BagelOrb, 2018-08-02:
The problem was most probably introduced in f06302c359ff3706c5ec3dd7d63061c42fb9e8a3 by @alekseisasin

BagelOrb, 2018-08-02:
Scrap those comments, I was looking at the wrong layer ^^

BagelOrb, 2018-08-02:
Oh this was already fixed. See https://github.com/Ultimaker/Cura/issues/4009. Fixed in https://github.com/Ultimaker/CuraEngine/commit/af07b7cee963953b64fc3eb7cddbd06b6222130a

closed 2018-08-02 by BagelOrb

printingotb, 2018-08-02:
Perfect, thanks! Didn't see that one.