Ultimaker/Cura #4166: Z position ignored when slicing

FynDestiny, 2018-07-31:
**Application Version**

4.3.1

**Platform**

Windows 10

**Printer**

[CCR10_balance.curaproject.curaproject.3mf.zip](https://github.com/Ultimaker/Cura/files/2247205/CCR10_balance.curaproject.curaproject.3mf.zip)

**Steps to Reproduce**

Add STL
change Z axis, in my case +10
Slice

**Actual Results**

Solid View, after slicing, shows item in expected position, Z10.
[image]

Layer View shows item starting a Z0
[image]

**Expected results**

Layer View should show item starting a Z10
This pic (taken before slicing), shows where I expect the Layer View should show the item.
[image]

**Additional Information**

 I want to print one item on top of another, using a different colour.
Unfortunately my printer has only one hot-end, so I have to print in two or more stages to complete my print ( I don't want to glue items together if I can print one atop another ).
With the ability to position an item at a higher Z axis, as you can in Cura, makes my print possible.
Unfortunately I don't know enough about gcode to edit the file to correct this.

fieldOfView, 2018-07-31:
See the setting "Remove Empty First Layers" in the Mesh Fixes category.

closed 2018-08-01 by ianpaschal

FynDestiny, 2018-08-01:
Did as you suggested. But it did not slice all of the item, it missed off the top few laters

Ghostkeeper, 2018-09-25:
The missing layers at the top have been fixed in 3.5.

fieldOfView, 2018-09-25:
Was this fixed post-beta, or should the fix be included in the beta?