Ultimaker/Cura #4167: Cura Inserting a gap in solid top/bottom layers

raiyneofgailin, 2018-07-31:
**Application Version**
3.4.1

**Platform**
Don't think it's related, but Windows 10.

**Printer**
Doesn't apply, but CR-10S (standard profile that comes with Cura latest version).

**Steps to Reproduce**
When I bring in a new STL design file, Cura is automatically inserting a gap in the center of the top and bottom layers (yes, the gaps end up getting included in the print). Simply import the STL I've attached into Cura v3.4.1, and look at the bottom layer as well as the top side of that horizontal segment and notice it's inserted a gap right down the center. This has happened on other files as well.

[gapissue.zip](https://github.com/Ultimaker/Cura/files/2247442/gapissue.zip)

**Actual Results**
It obviously inserted a big gap in my layers.

**Expected results**
It obviously SHOULDN'T insert gaps when those gaps didn't exist in the original file.

**Additional Information**
I already included a screenshot of Cura in the zipped file.

smartavionics, 2018-08-01:
That's possible when you use the concentric line skin pattern as the area may not be exactly an integer multiple of line widths wide. Try using Lines or Zig-Zag, patterns, they don't leave a gap.

breign, 2018-08-01:
This gap is a bug. Cura adds some gapping in circular plates. Print is useless.

[image]
[image]

[Vlozek-freecad-export.stl.zip](https://github.com/Ultimaker/Cura/files/2248960/Vlozek-freecad-export.stl.zip)

smartavionics, 2018-08-01:
To get rid of the gap, slightly increase/decrease the skin line width so that an even integer multiple of lines fit into the width of the skin area.

breign, 2018-08-01:
@smartavionics changing the concentric pattern to lines or zig-zag is the solution that works for me.

closed 2018-08-01 by ianpaschal

raiyneofgailin, 2018-08-01:
@breign and @smartavionics THANK YOU! Changing the pattern to lines solved the issue. Not sure why that wasn't set as the default pattern...