Ultimaker/Cura #4168: Windows: cura dont save files anymore

nehu92, 2018-08-01:
I have been trying the new version of cura 4.3.1 and really dont save files, i have try it in diferents versions of windows 10, with the same result. i also checked this isssue also has already happend in Ubuntu.

**Application Version**
4.3.1
**Platform**
Windows 10
**Printer**
custom
**Steps to Reproduce**
1. open or create file
2. save project...
3. save
4. check in file manager
or:

1. slice project
2. save to external SD
3. check in file manager - find a 0byte file

**Actual Results**
A 0kb file was created

**Expected results**
a complete gcode saved in sd

**Additional Information**
.log of the error
```
2018-07-31 21:37:52,795 - ERROR - [MainThread] UM.Logger.logException [81]: Exception: Unable to write to file moon4inch.stl: zero-size array to reduction operation minimum which has no identity
2018-07-31 21:37:52,801 - ERROR - [MainThread] UM.Logger.logException [85]: Traceback (most recent call last):
2018-07-31 21:37:52,807 - ERROR - [MainThread] UM.Logger.logException [85]:   File "X:\3.4\build\inst\lib\python3.5\site-packages\UM\Qt\Bindings\OutputDeviceManagerProxy.py", line 126, in _writeToDevice
2018-07-31 21:37:52,812 - ERROR - [MainThread] UM.Logger.logException [85]:   File "C:\Program Files\Ultimaker Cura 3.4\plugins\LocalFileOutputDevice\LocalFileOutputDevice.py", line 131, in requestWrite
2018-07-31 21:37:52,818 - ERROR - [MainThread] UM.Logger.logException [85]:     self.writeStarted.emit(self)
2018-07-31 21:37:52,824 - ERROR - [MainThread] UM.Logger.logException [85]:   File "X:\3.4\build\inst\lib\python3.5\site-packages\UM\Signal.py", line 212, in emit
2018-07-31 21:37:52,827 - ERROR - [MainThread] UM.Logger.logException [85]:   File "X:\3.4\build\inst\lib\python3.5\site-packages\UM\Signal.py", line 319, in __performEmit
2018-07-31 21:37:52,833 - ERROR - [MainThread] UM.Logger.logException [85]:   File "X:\3.4\build\inst\lib\python3.5\site-packages\UM\Signal.py", line 212, in emit
2018-07-31 21:37:52,838 - ERROR - [MainThread] UM.Logger.logException [85]:   File "X:\3.4\build\inst\lib\python3.5\site-packages\UM\Signal.py", line 315, in __performEmit
2018-07-31 21:37:52,844 - ERROR - [MainThread] UM.Logger.logException [85]:   File "C:\Program Files\Ultimaker Cura 3.4\plugins\UFPWriter\UFPWriter.py", line 26, in _createSnapshot
2018-07-31 21:37:52,850 - ERROR - [MainThread] UM.Logger.logException [85]:     self._snapshot = Snapshot.snapshot(width = 300, height = 300)
2018-07-31 21:37:52,856 - ERROR - [MainThread] UM.Logger.logException [85]:   File "X:\3.4\build\inst\lib\python3.5\site-packages\cura\Snapshot.py", line 89, in snapshot
2018-07-31 21:37:52,862 - ERROR - [MainThread] UM.Logger.logException [85]:   File "X:\3.4\build\inst\lib\python3.5\site-packages\cura\Snapshot.py", line 28, in getImageBoundaries
2018-07-31 21:37:52,867 - ERROR - [MainThread] UM.Logger.logException [85]:   File "numpy\core\fromnumeric.py", line 2420, in amin
2018-07-31 21:37:52,872 - ERROR - [MainThread] UM.Logger.logException [85]:   File "numpy\core\_methods.py", line 29, in _amin
2018-07-31 21:37:52,877 - ERROR - [MainThread] UM.Logger.logException [85]: ValueError: zero-size array to reduction operation minimum which has no identity
```

fieldOfView, 2018-08-01:
Duplicate of #4042

ianpaschal, 2018-08-01:
^Indeed.

closed 2018-08-01 by ianpaschal