Ultimaker/Cura #4173: Per Model Settings

FynDestiny, 2018-08-01:
**Application Version**

4.3.1

**Platform**

Windows 10

**Printer**

[CCR10_A Frame Back.curaproject.3mf.zip](https://github.com/Ultimaker/Cura/files/2250587/CCR10_A.Frame.Back.curaproject.3mf.zip)

**Steps to Reproduce**

In the main menu on the right of the display the generate support is not selected, when it is, all settings are default, (support pattern is zigzag).

Loaded up 4 parts.
Selected Print Sequence "One at a Time".
Added Generate Support and Support Pattern ( which would be nice if available in All at Once mode ) for each model.
Selected Generate Support and Pattern as "Lines" for one of the models.
Sliced.

**Actual Results**

Checked result.
The support pattern is Zig Zag and not Lines

**Expected results**

I was expecting the support pattern to be different from the default

**Additional Information**

[image]

smartavionics, 2018-08-01:
Looking into the cura source, it appears that the Support Pattern setting cannot be set on a per-mesh basis. It can be set either globally or per-extruder. I do not know how you have been able to specify it separately for the selected model in the image above (that's front-end black magic that I have no knowledge of). Hope this helps.

FynDestiny, 2018-08-01:
Sorry, not really.

If you have the Special Modes Print Sequence set to "All at Once", the option for Support Pattern in the Per Model Settings doesn't show. But it does in the "One at a Time" Print Sequence.

To be honest, having access to Support Pattern in any Print Sequence allows us to print multiple models that need different Support Patterns, for example, and sorry I didn't supply a photo of all the models in the project, one or two models require the Line pattern because it's a circular print and the others Concentric because they are square prints.

It would help in the fact that we'd have to load up the models and faff around trying to find which models require which Support Pattern, to create a multiple print.

smartavionics, 2018-08-01:
OK, that's all understood. Hmm, it doesn't sound unreasonable that you should be able to specify those settings for one at a time processing but I'm not familiar with the support generation part of the engine so I can't really say how difficult/easy/likely it will be for your suggestion to be implemented. Sorry.

FynDestiny, 2018-08-01:
Could you please pass this on to the support generation department, thanks

smartavionics, 2018-08-01:
Just leave the issue open and the Cura devs will see it and (hopefully) respond.

FynDestiny, 2018-08-01:
Cheers

printingotb, 2018-09-13:
Hi, wanted to add my observations.
I noticed similar issues when trying to ad a brim to two of four models on the buildplate (overall setting is skirt).
If I have "One at a time" selected, I can choose "buildplate adhesion type" and number of brim lines etc. But my changes here don't actually change anything when printing.

If I change to "all at once", the per model brim settings I choose disappear and when I enter the menu ta add them back, they are not selectable.

Not sure if we are supposed to be able to choose adhesion type for different models or there is a bug allowing us to select them when using "one at a time".

Ghostkeeper, 2018-11-13:
So here's what happened.

CuraEngine has a notion of a "meshgroup": A group of meshes that get printed together from layer 0 to layer N. When you select one-at-a-time mode, every mesh gets put in such a mesh group so that they get printed from layer 0 to layer N separately. For these mesh groups, it is possible to change a lot more settings than would otherwise be possible for single meshes, because CuraEngine effectively repeats the entire slice process for each mesh group separately, and this ensures that support doesn't overlap, etc.

Around March this year, it was decided to then allow many more settings to be added to the per-mesh setting dialogue if printing in one-at-a-time mode. You could edit more settings then. Awesome!

However in October we discovered that these settings actually never got used properly because they are overwritten by the per-extruder settings. Most settings get overwritten by the extruder, so changing a per-meshgroup setting did not have any effect. So now we removed these per-meshgroup settings from the menu again. You cannot edit them any more in 3.6, even if you set it to one-at-a-time mode.

Sorry for the confusion!

closed 2018-11-13 by Ghostkeeper