Ultimaker/Cura #4175: [3.4.1] Infill speed ignored

mbmosher, 2018-08-02:
**Application Version**

3.4.1
**Platform**

OSX
**Printer**

Ender3
**Steps to Reproduce**

Adjust infill speed to match inner wall speed
**Actual Results**

Does not appear to reduce infill speed
**Expected results**

Expect slower infill speed
**Additional Information**

Tried reduceing infill speed from 80 to 40, but this appears to have no affect on actual infill speed.  Project, gcode, and screen shots attached.

[Archive.zip](https://github.com/Ultimaker/Cura/files/2252558/Archive.zip)

BagelOrb, 2018-08-02:
@mbmosher you have attached the 3mf file, but we need to .curaproject.3mf file. Your file does not include the settings you have used.

Rabufs, 2018-08-20:
I think I am having the same problem.   I set my print speeds to be slow however the infill is going super fast.  I can't get the infills to slow down.  I attached my files that may help. I included the curaproject.3mf file too....    After a couple of ruined prints, I started using the Cura layer view and press play at one of the infill layers.  The nozzle zips almost instantly along an infill line but when it gets to a curved line on the wall, it slows right down...  The infill is extremely fast...
I don't normally set the infill print speed, but I checked it after finding out about it.  It was defaulted to 100.  I tried to over-ride to a slower speed to no avail.
[Fast-infill.zip](https://github.com/Ultimaker/Cura/files/2304232/Fast-infill.zip)

Rabufs, 2018-08-21:
I also happened to noticed that when the infill speed has the reset icon (and 60), and I press reset, it goes to what it says is an absolute value (and the "i" symbol) with a speed of 100.  When I click the i, it goes to 60 and shows the reset icon.   I always have "reset" or "i".  If if I type in a slower speed such as 16 then I see both i and reset....  I'm not sure if this helps to know.
Just to be clear, even if I type in a slow speed such as 16, the infill print speed is super fast...

[image]
[image]

BagelOrb, 2018-08-21:
The layer view is not time based. All moves are instant.

This is probably a default setting in your printer profile.

Please attach a project file so that we can reproduce it.

Rabufs, 2018-08-23:
So are these files sufficient to see what the problem is [Fast-infill.zip](https://github.com/Ultimaker/Cura/files/2312448/Fast-infill.zip)?  It does have Sample_base.curaproject.3mf in the zip file.  Do you need anything else?

Rabufs, 2018-08-24:
Hello?

Rabufs, 2018-08-25:
Uninstalling Cura alone does not resolve the problem.  I uninstalled and reinstalled it a few times to no avail.  I uninstalled Cura again and also deleted all Cura folders in C:\Users\your user ID\AppData\Roaming and I think it was also in C:\Users\your user ID\AppData\Local.    You may need to adjust your Windows setting to allow you to view hidden folders.   I then installed Cura and it seems to be working better for now...  Before you do this, just be prepared to redo your initial configuration again.

fieldOfView, 2018-08-26:
Reinstalling Cura only ever fixes something if you have manually edited files in the application bundle. Normal configuration changes only happen in the configuration folder.

Rabufs, 2018-08-26:
The only changes I ever made was setting temperatures and speeds for PLA and PETG.  I didn't even know there was an infill speed until I started having problems.  ...and when I found it, the infill speed made no difference...    Before the problem started,  I tried to save my changes once or twice (I think hit keep my changes) but Cura kept crashing and restarting / initializing.  It may be related to that...   No worries.  I'm not sure what caused the infill speed to change the way it did but it seems to be OK now.  Thanks.

BagelOrb, 2018-08-29:
I generated several gcodes from the attached project with Cura 3.4.1: with infill speed set to 8, 16, 40 and 80.

All gcodes contain the expected speeds (in mm/min rather than mm/s) in the gcode.

e.g.
```
;LAYER:6
G0 X127.269 Y127.269 Z1.1
M204 S500
M205 X20 Y20
;TYPE:FILL
G1 F2400 X172.743 Y172.743 E1488.3761
G1 X172.813 Y172.813
M204 S5000
M205 X30 Y30
G0 F7200 X167.644 Y176.875
M204 S500
M205 X20 Y20
G1 F2400 X178.725 Y135.52 E1489.4975
```

I therefore conclude that this problem is a problem of your machine - not of Cura.

closed 2018-08-29 by BagelOrb