Ultimaker/Cura #4180: 3.4.1  can not change the settings like " print speed" , "Infill density" and some others.

zj-yuan, 2018-08-03:
**Application Version**

Cure 3.4.1
**Platform**

Win8.1 Chinese
**Printer**

**Steps to Reproduce**

**Actual Results**

**Expected results**

**Additional Information**

3.4.1  can not change the settings like " print speed" , "Infill density" and some others.

ianpaschal, 2018-08-03:
Please be more descriptive. What are you seeing? Are the boxes blank? Do they have a value that can't be edited? Do they accept new values but always revert?

We can do nothing without detailed information about what's happening.

BagelOrb, 2018-08-04:
If you set he infill line distance the infill percentage cannot be set anymore. That's by design because the only think the density controls is the line distance.

Similarly if you set all the speeds then the print speed doesn't do anything any more so the setting is disabled.

Ghostkeeper, 2018-09-28:
Let's assume that BagelOrb's suggestion was the issue that zj-yuan was having. It's working as intended then. Feel free to correct us if you're still having this issue.

closed 2018-09-28 by Ghostkeeper