Ultimaker/Cura #4183: [3.4.1] Full build volume not usable

r3turnz, 2018-08-03:
**Application Version**
3.4.1

**Platform**
Windows

**Printer**
Ultimaker S5

**Steps to Reproduce**
Attempt to slice a model with a diameter of 237mm. Build plate adhension must be disabled.

**Actual Results**
Cura reports exceeded build volume.

**Expected results**
Slicing process completes in version 3.3.1.

ianpaschal, 2018-08-03:
This is to be expected. You also need to set "travel avoid distance" to zero. This is an additional buffer so that the print head can travel around printed parts without crashing into them. I don't recommend setting it to zero, but if you do, you can fit a 237 mm object on the build plate.

r3turnz, 2018-08-03:
Altough I have set the "travel avoid distance" to zero, the print still doesn't fit on the build plate. The required space is still occupied by a shadow. 
[image]

ianpaschal, 2018-08-03:
Strange. Are you sure your object is 237 mm?

3.4.1 works fine on my computer.

[image]

ianpaschal, 2018-08-06:
I really can't reproduce this. Make sure build plate adhesion is off, and "avoid printed parts when traveling" is turned off for both extruders, and it should work.

Unfortunately there's little we can do if we can't reproduce the bug. I'm leaving this open in case someone else is able to.

Also, if you could upload a `.curaproject` file with the bug, we can try to investigate it that way.

r3turnz, 2018-08-07:
I found the reason, why you can't reproduce this. The model must require support structures and the secondary extruder must print these.

closed 2018-10-10 by Ghostkeeper