Ultimaker/Cura #4191: Brim line count causes incorrect outlines

agardip, 2018-08-04:
**Application Version**

3.4.1
**Platform**

Windows 7 Professional 64bit
**Printer**

BQ Hephestos2
**Steps to Reproduce**

Just set Brim line count bigger than 2
**Actual Results**

Outline errors
Setting brim line count to 3-4 or above results in huge numbers of x and y in g code, for example:

G1 X61.884 Y121.426 E411.25033
G1 X61.823 Y125.676 E411.46239
G1 X92.575 Y193.549 E415.17994
M204 S5000
G1 F1500 E412.17994
G1 F300 Z1.3
G0 F2400 X99871.296 Y647461.956 Z1.3  <----
M204 S1000
G1 F300 Z0.3
G1 F1500 E415.17994
G1 F1200 X62.901 Y112.023 E33093.21126
G1 X62.528 Y116.34 E33093.42744
G1 X62.284 Y121.45 E33093.68267
G1 X62.229 Y125.276 E33093.87357
G1 X147.767 Y125.277 E33098.14108
G1 X147.707 Y121.066 E33098.35119
G1 X147.475 Y116.43 E33098.58277
G1 X147.158 Y112.625 E33098.77326
G1 X146.758 Y109.275 E33098.94158
G1 X99871.296 Y647461.956 E65776.47116   <----
M204 S5000
G1 F1500 E65773.47116
G1 F300 Z1.3
G0 F2400 X61.245 Y155.86 Z1.3
M204 S1000
G1 F300 Z0.3
G1 F1500 E65776.47116
G92 E0
G1 F1200 X146.403 Y109.675 E4.83316
G1 X63.592 Y109.676 E8.96462
G1 X63.3 Y112.056 E9.08424
G1 X62.924 Y116.412 E9.30237

**Expected results**

Nice outlines as brim
**Additional Information**

The error is random, if brim line count set to say 6, slice, then set to 7, slice, then back to 6, the result is different.
[BQH2_nagyalaktreszek.curaproject.3mf.zip](https://github.com/Ultimaker/Cura/files/2259463/BQH2_nagyalaktreszek.curaproject.3mf.zip)

BagelOrb, 2018-08-04:
This problem has already been solved for the next release.

I quick search through he existing issues can give you more info if you would like that.

You can prevent this behaviour by enabling the setting Brim on Outside Only

closed 2018-08-04 by BagelOrb

agardip, 2018-08-04:
Thanks BagelOrb!

Yes, I usually use Brim on Outside Only, but previous components needed to have this unchecked. Will use it for this one.