Ultimaker/Cura #4192: "Arrange All Models" in "One at a Time" print sequence

FynDestiny, 2018-08-04:
**Application Version**

4.3.1

**Platform**

Windows 10

**Printer**

Creality CR-10S
[CCR10_balance.curaproject.3mf.zip](https://github.com/Ultimaker/Cura/files/2259714/CCR10_balance.curaproject.3mf.zip)

**Steps to Reproduce**

First 4 photos show sequence
Load model
Duplicate model to 3
Switch to "One at a Time" print sequence
Select "Arrange All Models"

**Actual Results**

Fifth photo shows result
Two models positioned off build plate

**Expected results**

Sixth photo shows expected arrangement
Sixth to Eighth photos show XY positions of models

I understand that rotating a model to fit might make "Arrange All Models" more complicated to program, but using just the max width and height of the models required print clearance, should make it easier to calculate.

**Additional Information**

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

ChrisTerBeke, 2018-08-04:
The grey area around the object indicates the nozzle can't go there (because of one-at-a-time mode which requires clearance between the objects so that the extruder doesn't collide with them). Adding rotation to auto-arrange makes it a mathematical problem of a completely different category...

fieldOfView, 2018-08-04:
Note that there is no such thing (yet) as Cura 4.3.1. You probably mean Cura 3.4.1.

FynDestiny, 2018-08-04:
Yes, I meant 3.4.1 . Slip of the digits.

FynDestiny, 2018-08-04:
I understand adding the rotation would complicate matters.

But even without the rotation, two of the three models can be positioned perfectly well in the build area. As you can see in the last few photos. Leaving just the one in need of "adjustment" by the user, to either make it fit or remove it

Ghostkeeper, 2018-09-28:
There are two collision areas in one-at-a-time mode. The inner area of one object is not allowed to intersect with the outer area of the other object. The arranger works by painting a pixel grid of areas that are filled with models. Ideally, when checking whether an object is allowed to be placed there it should use a different area than when filling the pixels that are obstructed. This would pack these objects much more efficiently.

nallath, 2018-10-01:
Except that it would need to use 2 different areas (that's what the push free is doing). The arranger is totally not built to handle this at the moment.

So unfortunately, this is an issue, we should fix it at some point, but it's not something that's used all that often, so we probably wont.

Bob90, 2021-03-15:
Just want to add that the arranger also sets the models in from the edges, but you would normally be able to print there so the models along an edge could be arranged much further out and provide more space.

RandomGgames, 2022-07-29:
Wow this hasn't been improved in 4 years. Could we get this added? There are many instances where there are a lot more models that could fit onto a plate by manually placing them. The auto arrange works great in normal mode but not in one at a time... And I'm using 4.13.1!!

behold81, 2023-01-04:
This is STILL an issue!! 

Surely there are a few things that could be taken in to account to solve it. Also, the grey areas can overlap the edge of the build area as this is dead space.