Ultimaker/Cura #4193: "Support Blocker" preventing Slicing in "One at a Time" Print Sequence

FynDestiny, 2018-08-04:
**Application Version**

4.3.1

**Platform**

Windows 10

**Printer**

Creality CR-10S

**Steps to Reproduce**

Load multiple models.
Set Support Blocker where desired.
Switch to "One at a Time" Print Sequence.
Slice.

**Actual Results**

Slicing starts, gets about half way through, then appears to "crash".
The progress bar clears itself, but Cura appears to be still slicing, and remains this way until "Cancel" is clicked.
[image]
[image]

**Expected results**

Expected results are for the slicing to complete with the support missing from the blocked areas.

**Additional Information**

This arrangement of settings works fine in the "All at Once" Print Sequence.
But I require a multiple print done, one model at a time, but with limited support.

Alternatively, a third option in "Support Placement" of "On Model Only", allowing support material to be "everywhere needed on the model" except on the print bed

fieldOfView, 2018-08-08:
Please add a project file (file -> save project)

FynDestiny, 2018-08-09:
Hi.
Since writing this I've changed to a different project and this issue doesn't appear to be happening this time.
If it should repeat this issue, I will update you.

FynDestiny, 2018-08-09:
Though my idea of a third option of "On Model Only" would make some prints easier, where support is only wanted for an overhang that is only over the model and not the printbed

Ghostkeeper, 2018-10-11:
I don't really see an advantage of supporting only when above model, and not on the build plate. What case would need something like that? Support resting on the build plate should be fine.

FynDestiny, 2018-10-11:
What is the point in having so many print options when they are not all available in both print modes. Same issue with individual model settings, you should be able to print any number of models (space dependent of course), with any of the settings, even different layer thickness if wanted.

Cura is good, easy to use, but lacks total flexibility.

fieldOfView, 2018-10-11:
I am not suggesting that the current situation is ideal, I just came to the realisation that you uncovered a bug that I don't think the developers have thought about before. When you print "on at a time", Cura will currently try to print the main model first, and then the support blockers one by one, because those are seen by Cura as individual models too. At least, I think they are.

GregValiant, 2024-11-14:
Is this still an issue with current versions of Cura (5.8.0 and up)?  Can this be closed?

closed 2024-12-10 by github-actions[bot]