Ultimaker/Cura #4194: "Center selected models" of merged model centers on "average center" not "geometric center"

BenoitValin, 2018-08-05:
**Application Version**
3.4.1

**Platform**
OS X

**Printer**
UM3x
[Center Selected Models.3mf.zip](https://github.com/Ultimaker/Cura/files/2260009/Center.Selected.Models.3mf.zip)

**Steps to Reproduce**
1. Add two models of an assembly
2. Merge 
3. "Center selected models" 

**Actual Results**
Assembly is centered around the average center of the multiple models that were merged together  and not the geometric center of the assembly, as when "Center selected models" are done on individual parts.
[image]

**Expected results**
The whole assembly should be located in the geometric center of the build plate
[image]

**Additional Information**
nothing

diegopradogesto, 2018-08-06:
Thanks for reporting. Sounds like a bug to me, but it's happening for a long time. So maybe in the past it was decided that this is the right behavior. I will discuss with the colleagues.

diegopradogesto, 2018-08-06:
Devs. CURA-5617

ianpaschal, 2018-08-06:
Just to be clear, this is a bit confusing how the internal ticket is written. Currently it appears to move the center towards the center of mass (i.e. the side of the white piece which has the extra bit on it) when in fact it should be moved to the center of the resulting bounding box.

Using the bounding box will ensure that this works for this scenario as well as when two objects are not overlapping.

Also the described behavior refers to grouped models, not merged models, which both snap to the center (probably not the intended behavior but also a separate bug.

closed 2018-08-09 by ianpaschal