Ultimaker/Cura #4195: [3.4.1] Minor issue/Potential user error: Cura can be closed while gcode is written -> incomplete file

Cyberman-tM, 2018-08-05:
**Application Version**
3.4.1

**Platform**
Windows 10 64Bit

**Steps to Reproduce**
*) Slice a file
*) save gcode
*) immediately close cura (X at top right)
*) check gcode file

**Actual Results**
The gcode file is cut off near the end, after ";TIME_ELAPSED:"

**Expected results**
Cura should either complete the write process, or prevent the user from closing the program while it writes.
(Please don't add a constant "do you really wish to close?" question :-)

ChrisTerBeke, 2018-08-05:
It should indeed not close before an action is finished (like writing G-code).

Cyberman-tM, 2018-08-05:
I *think* the problem might be that the current action IS completed - the action of writing the actual core-gCode - the next action would be finishing up the file, adding the closing gcode, the parameters, etc...

ChrisTerBeke, 2018-08-05:
I'd say all of that is part of 1 export 'action'.

alekseisasin, 2018-08-07:
@Cyberman-tM , We have discussed your request and unfortunately we are not going to make changes in exporting/saving Gcode process because we already have a message box with a progress bar which indicates whether a file is ready or not.

ChrisTerBeke, 2018-08-07:
@alekseisasin can you update the ticket labels and open status then?

Cyberman-tM, 2018-08-07:
@alekseisasin: No problem, as I said, it's basically a user error. It's just something I thought worth mentioning, maybe someone else encounters this situation and finds this. I'll close the ticket.

closed 2018-08-07 by Cyberman-tM

alekseisasin, 2018-08-07:
@ChrisTerBeke , I do not close immediately, I have a sticky on my table, to check reply from a person who created the ticket. I saw a couple of times opening and closing tickets because of a misunderstanding.
@Cyberman-tM , thank you for your idea and closing the ticket.

Ghostkeeper, 2018-10-10:
This was fixed in 3.5.