Ultimaker/Cura #4196: Cura 3.4.1 will not connect to Ender 3 via USB

MikeUSNRet, 2018-08-05:
**Application Version**

**Platform**

**Printer**

**Steps to Reproduce**

**Actual Results**

**Expected results**

**Additional Information**

MikeUSNRet, 2018-08-05:
Thanks.  Thought that this would be a good place to start since it worked before flashing a new version of marlin firmware to the printer and other 3rd party programs do connect to the printer via USB.  was on the thought process that this was an incompatability issue with Cura and Marlin 1.1.9.

ChrisTerBeke, 2018-08-05:
It would help if you can attach some logs (cura.log and maybe some communication logs from the USB port). That would help people here to find the problem.

JohnEdwa, 2018-08-05:
I have an Ender 3 currently running TH3D U1.R1.9b and a laptop with Win 7 64bit which is able to successfully connect to Cura, and I've printed multiple objects with it as well, so I can at least say it's not a problem with the firmware and Cura being incompatible with each other. It could be a Win 10 issue, but it's more likely that there is something wrong with your system specifically.

Ollowitsch, 2018-08-11:
I had exactly the same problem with a CR-10 after updatig to new Marlin-version and posted my solution also in the [ultimaker community](https://community.ultimaker.com/topic/24063-cura-341-usb-select-mini-pro-not-working/)

The problem:

Cura 3.4.1 (actually it's plugin "USBPrinting") is sending gcode "M105" (request temperature) at different baudrates to the serial port. It expects at least 3 successfull response lines containing the string "ok T:" - but new Marlin is only sending 1 line matching this string - containing all temps (i.e. "ok T:27.03 /0.00 B:26.91 /0.00 @:0 B@:0\n")

The solution:

Change file C:\Program Files\Ultimaker Cura 3.4\plugins\USBPrinting\AutoDetectBaudJob.py (or whereever your Cura Plugin-dir is located) line 63 from:

if successful_responses >= **3**:

...to:

if successful_responses >= **1**:

...and restart Cura, which will now find the correct baudrate (propably 115200) and enable the serial communication

wjohnsaunders, 2018-10-08:
There is a comment in AutoDetectBaudJob.py that says the following:
# Ensure that we are not talking to the boot loader. 1.5 seconds seems to be the magic number

It seems Marlin 1.1.9 on the Ender 3 is happier with a 5 second delay. I also removed all baud rates, except 115200, from the list all_baud_rates. As they are not used and only makes a connection failure worse.

It seems having an Arduino bootloader resident on the printer causes Marlin to reset each time the com port is opened. I have also tested with with Putty with no serial handshaking enabled.

cds333, 2018-11-30:
Confirmed that changing AutoDetectBaudJob.py lines:

def run(self):
wait_response_timeouts = [3, 15, 30]

TO:

def run(self):
wait_response_timeouts = [5, 15, 30]

Fixed the problem for me

sam-ward, 2018-12-27:
I just wanted to add that I was having the same problem with my Ender 3 in Cura 3.6, but updating the wait_response_timeouts as above fixed it for me.

fluxkompensatorausderzukunft, 2018-12-30:
> Change file C:\Program Files\Ultimaker Cura 3.4\plugins\USBPrinting\AutoDetectBaudJob.py (or whereever your Cura Plugin-dir is located) line 63 from:
>
> if successful_responses >= **3**:
>
> ...to:
>
> if [...]

This fixed it for me an my Anet A6 - thanks a lot!

Lijcraft, 2019-01-02:
I'm new at 3d printing and I cant change it, when I hit save it says something about my access being limited or denied or something, when I try to change the autodetectbaudjob.py file.  Am i missing something?

Windows 10
cura 3.6

tcarlisle2012, 2019-01-02:
> Also, im using notepad to edit it.

You have to have admin privileges to change files under C:\Program
Files\Ultimaker Cura 3.4

The default permissions on a windows system does not allow normal users to
write to the program binaries paths, and that is a good practice actually
as it helps minimize damage if you get bitten by something malicious.
Please don't change this. Instead, either log in as the admin, or run
notepad elevated as admin. You can do this by typing "notepad" in the
windows search bar, and when notepad pops up don't left click it, and
instead right-click it and select "Open as Administrator".

Of course, you need to know an admin account and password to do this. If
you cannot, then whoever installed Cura on the system in the first place
will have to help you out.

Lijcraft, 2019-01-02:
> You have to have admin privileges to change files under C:\Program
Files\Ultimaker Cura 3.4

The default permissions on a windows system does not allow normal users to
write to the program binaries paths, and that is a good practice actually
as it helps minimize damage if you get bitten by something malicious.
Please don't change this. Instead, either log in as the admin, or run
notepad elevated as admin. You can do this by typing "notepad" in the
windows search bar, and when notepad pops up don't left click it, and
instead right-click it and select "Open as Administrator".

Of course, you need to know an admin account and password to do this. If
you cannot, then whoever installed Cura on the system in the first place
will have to help you out.

I did this, now do I change one at a time or both, (Sorry if this is a stupid question)

tcarlisle2012, 2019-01-02:
There is only one change to make, in that one file:  See instructions
above, find line 63 and make the change as described, then save the file.
It wouldn't be a bad idea to open the file again and make sure the change
was saved.

You might find notepad++ a little easier to work with, and you won't have
to worry about things like line endings, etc.

If you did the change, and verified it saved, and have completely shut down
cura and restarted, then this fix just doesn't happen to solve it for you.
But I would reboot just to be sure.

Nheters, 2019-01-27:
Hi, I'm using cura 3.6, Win10 and i have a creality ender 3, and i have the same exact behavior, where the printer doesnt connect to the Notebook, and Cura does not recognize it.

Ghostkeeper, 2019-02-13:
I think increasing the first response time-out from 3 to 5 seconds shouldn't be a problem. It'll increase the detection time by 2 seconds for some printers though.

nallath, 2019-02-13:
The number of sucesfull_responses needed has already been changed to 1 (instead of the old 3), so I don't think we need to change the timeout.

Ghostkeeper, 2019-02-14:
So if Ollowitsch and Fluxkompensatorausderzukunft are correct it should be fixed in version 4.0 then.

closed 2019-02-14 by Ghostkeeper

b1naryVap0r, 2019-05-12:
> Confirmed that changing AutoDetectBaudJob.py lines:
> 
> def run(self):
> wait_response_timeouts = [3, 15, 30]
> 
> TO:
> 
> def run(self):
> wait_response_timeouts = [5, 15, 30]
> 
> Fixed the problem for me

a few things:
First - cura is great, it is good of ultimaker to provide cura even though i cant afford their printers. That said this is good for the comunity and builds their brand which is cool. in the future i will own one of the Ultimaker printers- for now ill have to make do with a cheaper options and tinker to get passable prints.

its cool that this program is made in python. love python.

problem i had

ender 3 pro wasnt connecting to printer on version 3.6 on ender 3 pro- after flashing to TH3D_UFW_U1.R2.10d ::

 i took these 2 fixes from below --

in filepath where cura is installed adjust: AutoDetectBaudJob.py 

change:--
def run(self):
wait_response_timeouts = [3, 15, 30]

TO:

def run(self):
wait_response_timeouts = [5, 15, 30]

& &&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&

if successful_responses >= 3:

...to:

if successful_responses >= 1:

then restart Cura

Ghostkeeper, 2019-05-13:
It's a bit dangerous to change that last bit though. If the output happens to be a valid decimal number then we connect to it. There's a reason why it asks for 3 successful responses.

b1naryVap0r, 2019-05-25:
confirmed that this problem still exists, with cura 4.0, reinstalled to use creawesome mod and had to alter the autodetectbaudjob.py so my ender 3 pro would be recognized / detected by cura. everything else went fine with the mod but the printer wasnt being detected - only an option to save to file. i reinstalled without using the mod, then tried cura 4.1 beta and the issue still happened, then i reinstalled cura 3.6 and remembered that this was a thing i had to do to allow communication between ender 3 and cura over usb. 

just confimed after doing the same fix above that i mentioned above and installing the creawesome mod on cura 4.0 for my ender 3 pro to print over usb that cura can see the printer.

sidenote - im being verbose / pedantic above so people can hopefully find this if they have the same problem, im comfortable with python and familiar with windows 10 cura and the ender 3 pro by now, but maybe someone else isnt and this can help.

hopsephiroth, 2022-10-20:
Hello Everyone
I install Cura 5.1.1 for work with my ANET 3D PLUS on Windows 10, 3 years from I re-box the 3D printer by home change.
So I don't remember this issue, thank you all of you for this trhead.

I just change:
def run(self):
wait_response_timeouts = [3, 15, 30]
TO:
def run(self):
wait_response_timeouts = [5, 15, 30]

And leave juts 115200 (erase all others baudrates)

And My cura 5.1.1 detect my printer.

On the new version of the file AutoDetectBaudJob.py there is not a line with:
if successful_responses >= 3:
I can´t change this. but it works now.

So, for newer Cura users: the file path has change on windows 10:
C:\Program Files\Ultimaker Cura 5.1.1\share\cura\plugins\USBPrinting
The folder Plugins are now inside a folder called "share"

Now I have other problem: Y movement is inverted, if I try autohome the Y motor moves aways from limit switch.

I'll search the solution of this Issue over the forum.

Thank you everyone again, I'm so Happy

Have a great day