Ultimaker/Cura #4197: The version of Cura available in the Ubuntu Software app is old

mrjohnc, 2018-08-05:
The current version of Cura that is available through the Ubuntu Software app is very out of date (3.1.0-1 vs 3.4.1 on the Ultimaker website).

[image]

ChrisTerBeke, 2018-08-05:
Ultimaker doesn't maintain that so you'll have to contact the publisher directly.

closed 2018-08-05 by ChrisTerBeke

mrjohnc, 2018-08-06:
OK, so its Ubuntu who added it, not Ultimaker? 

Thanks

fieldOfView, 2018-08-06:
That is correct. Ultimaker only distributes the AppImage version of Cura for Linux distributions.

mrjohnc, 2018-08-06:
OK, thanks very much for the information

mrjohnc, 2018-08-06:
Ok, one last question, then I'm done I promise :) Would it be possible to add a function to Cura on Linux which looked for new versions and installed them from within Cura. So no matter which out of date source it was downloaded from it would still give an easy and obvious path for users to keep Cura up to date.

In practice this would mean a box popped up when you start Cura that says 'a new version of Cura is available' and then you can click a button to update

fieldOfView, 2018-08-06:
This is already in place in the official distribution, but I think whoever packaged Cura for Ubuntu has left out that functionality because they intend for the package manager to be used instead of the Cura-internal update check.

mrjohnc, 2018-08-06:
Ok, thanks, I'll find out who it was and ask them nicely to add it back in

mrjohnc, 2018-08-06:
I found them :) Its @thopiekar