Ultimaker/Cura #4201: Cura3.3, add printer lead to breakdown. Why?

Kun-Tian, 2018-08-06:
Cura3.3, I run the project and the application start up.
When I add a printer, if I chose "Custom FDM printer", it will break down immediately.
But, if I chose "Ultimaker ***", it will start up with no crash.

What is the reason? Thanks!

Kun-Tian, 2018-08-06:
Here is the log.
[cura.log](https://github.com/Ultimaker/Cura/files/2261951/cura.log)

Kun-Tian, 2018-08-07:
It is not installation application.
I use pycharm run the code(Branch3.3), not the installation.
Thanks!

ianpaschal, 2018-08-07:
I'm not able to recreate this issue. Do you have all the dependencies installed correctly?

Kun-Tian, 2018-08-07:
It is the official version. 
I didn't change any code. The code is from:
https://github.com/Ultimaker/Cura/tree/3.3  and  https://github.com/Ultimaker/Uranium/tree/3.3

I just combine Cura and Uranium together, the file of materials, quality, variants, or other config file is initial file, didn't been changed.

If I chose the printer of Ultimaker, such as Ultimaker3, it will installed correctly.

Please download the code  from the above URL, and have a try, it will be recreate.
Thanks!

fieldOfView, 2018-08-07:
Don’t combine Cura and Uranium in one folder. Instead use the PYTHONPATH variable to “include” uranium when running cura_app.py.

Why do you use the 3.3 branches instead of the 3.4 branches?

Kun-Tian, 2018-08-08:
Now I separate Cura and Uranium in two folder, and use the PYTHONPATH variable to include Uranium , it still crashed.
I use 3.3 because the develop environment(cura-build-environment3.3) was build months ago, the build environment process is not easy, so I still use 3.3.  I will try 3.4 later.

But how to figure this problem.

Thanks!

ianpaschal, 2018-08-08:
Unfortunately we cannot spend time trouble shooting this. Without seeing your whole dev environment and making sure that everything is set-up properly, we can only guess what might be wrong.

Also, since we are all running Cura, we are unable to reproduce the issue, and by definition it affects 0% of our end users running the distributable binaries.

I'm afraid this is one you'll need to troubleshoot yourself.

closed 2018-08-08 by ianpaschal