Ultimaker/Cura #4202: [14.07] Cannot Enable Plugins

Jasman506, 2018-08-06:
**Cura Version**
14.07

**Platform**
Windows 7

**Steps to Reproduce**
I have downloaded and attempted to install the Cura Orientation, Print Cost Calculator, and Uranium Example Tool plugins. The instructions I was given said to download the files, unzip the folder, and move all of the files into a folder inside of the Plugins folder within the Cura program files.

**Actual Results**
Although I followed the given instructions for installing plugins, they seem to not be enabled.

**Expected results**
As I understand, the installed plugin(s) should be shown in the window alongside the Pause At Height and Tweak at Z plugins

**Additional Information**
I also attempted to copy only the .py file outside of the folder, as it seemed that might have made a difference (it didn't). I also noticed that the Pronterface UI plugin, which came pre-installed with the software, also was not enabled. Attatched is a screenshot documenting that, although the plugins are installed, they are not listed as enabled within Cura.
[image]

fieldOfView, 2018-08-06:
You are using an ancient version of Cura (4 years old) that is no longer supported. The plugins you are trying to use are for Cura 3 and newer.

ChrisTerBeke, 2018-08-06:
We don't support legacy Cura here anymore.

closed 2018-08-06 by ChrisTerBeke

ianpaschal, 2018-08-07:
To be more specific, we only support Cura version 2.7 and higher. For legacy Cura please see this repository: https://github.com/daid/LegacyCura