Ultimaker/Cura #4204: Cura add temperature update gcode when tool change, without need it.

mtmello2003, 2018-08-06:
Hello,

My case and for sure the same for others that uses a SINGLE NOZZLE hotend: I do not need any gcode to adjust temperatures after a tool change but cura does it and there isn´t  a way to avoid this behavior.

I think this one could solve that situation: https://github.com/Ultimaker/Cura/issues/3547

As a workaround I always need to run a post script to remove the cura adding M109 Sxxx gcode... but this remove also the first M109 Sxxx that I need to be there then I need to edit the file and activate manually the first M109...

Thank you.

karabas, 2018-10-08:
ChrisTerBeke commented on Mar 24:
"This will be added in the next version of Cura."
No settings in 3.5. Ok M109 removed but M104 Tx S0 on last extruder change still here.

ChrisTerBeke, 2019-02-18:
That was my answer because the question was apparently vague enough for me to interpret it as a request for a feature to disable one extruder. This was indeed added to Cura in the release after I made that comment.

Ghostkeeper, 2019-05-29:
Some experiments have been performed to work towards a solution for this issue, discussed in this ticket: https://github.com/Ultimaker/Cura/issues/5770

Dj-Garfield, 2020-01-24:
Hi All , I found a workaround to avoid Hotend temperature turning OFF , I just added this in all of my extruder Start GCODE :
M104 S{printing_temperature} , and the temperature didn't goes down at every tool changes.
Before this

mahtDFR, 2020-01-24:
This will also be implemented for the upcoming release (4.5)

closed 2020-01-24 by mahtDFR