Ultimaker/Cura #4205: Raft vs Brim

alittleticked, 2018-08-07:
**Application Version**
3.4.1

**Platform**
Windows 7 Ultimate w/Service Pack 1

**Printer**
CR-10 300 x 300 x 400

**Steps to Reproduce**
1. Be sure that an A8 printer is already configured. (Just to duplicate my setup)
2. Load a generic CR10 printer
3. Select 'Draft' quality - I tried them all, and they all did it.
4. Select 'Brim', 250 min length, 8.0 width, 20 line count, only on outside.
4. Make no other changes
5. Print anything.

**Actual Results**
The printer begins to print a raft, not a brim. This has happened after two months of use where brims were printed, now all of a sudden, no matter what object I print, I get a raft!

**Expected results**
I expect a brim
[image]

ianpaschal, 2018-08-07:
In addition, is this issue visible within the g-code/layer view? Or only when printing? If it's only when printing I'm afraid we won't be able to work on it as we can only handle bugs within Cura or on Ultimaker machines (we don't have a CR-10 in the office).

Ghostkeeper, 2018-10-10:
We cannot reproduce this. It produces a proper brim.

[image]

Since we have no further information on how to reproduce the issue, we can't fix it. We'll close it now then.

closed 2018-10-10 by Ghostkeeper