Ultimaker/Cura #4206: Having installed Cura 3.4.1 it's not possible to save files. After correct slicing either no files ar stored (on hard disk) or 0 bite file named correctly are stored

nosailor, 2018-08-07:
**Application Version**

**Platform**

**Printer**

**Steps to Reproduce**

**Actual Results**

**Expected results**

**Additional Information**

ianpaschal, 2018-08-07:
Well, it works for us, so please fill out the issue template and provide screen shots and project files which recreate the issue.

nosailor, 2018-08-07:
App-Version: Cura 3.4.1 (V. 3.2 was reinstalled by V. 3.4.1)
Plattf.: WIN 10 Prof
Printer: Anycubic I3 Mega
Steps: 
1. double-click on "stl"-file - (or pull in a stl-file into the Cura-Working Cube)
Result: Cura is starting correctly, the model is displayed and slicing is done automatically and (if SD-card is present) "save to removable disk" is offert.
2. activate "save to removable disk" 
Result:  saving to removable disk is indicated, moving the "load-bar" for about 30sec (Pic. 1) and after that  a gcode-file is produced with the correct name but with 0 content. (Pic. 2)
3. activate "save to file":
Result: The correct file-folder opens indicating the correct name at the bottom but no file is stored (Pic.3)

This behaviour is reproduceable.
Cura Version 3.2 was installed than again and is working perfectly
[image]
[image]
[image]

nosailor, 2018-08-09:
how about an answer

nosailor, 2018-08-09:
> It works fine here (on Mac though). Are you sure the disk is writable?
>
> [
> {
> "@type": "EmailMessage",
> "@type": "ViewAction",
> "target": "https://github.com/Ultimaker/Cura/issues/4206#issuecomment-411868819",
> "url": "https://github.com/Ultimaker/Cura/issues/4206#issuecomment-411868819",
> "name": "View Issue"
> },
> "description": "View this Issue on GitHub",
> "publisher": {
> "@type": "Organization",
> "name": "GitHub",
> "url": "https://github.com"
> }
> },
> {
> "@type": "MessageCard",
> "@context": "http://schema.org/extensions",
> "hideOriginalBody": "false",
> "originator": "AF6C5A86-E920-430C-9C59-A73278B5EFEB",
> "title": "Re: [Ultimaker/Cura] [3.4.1] It's not possible to save files. (#4206)",
> "sections": [
> {
> "text": "",
> "activityTitle": "**ChrisTerBeke**",
> "activityImage": "https://assets-cdn.github.com/images/email/message_cards/avatar.png",
> "activitySubtitle": "@ChrisTerBeke",
> "facts": [
>
> ]
> }
> ],
> {
> "name": "Add a comment",
> "@type": "ActionCard",
> "inputs": [
> {
> "isMultiLine": true,
> "@type": "TextInput",
> "id": "IssueComment",
> "isRequired": false
> }
> ],
> "actions": [
> {
> "name": "Comment",
> "@type": "HttpPOST",
> "target": "https://api.github.com",
> "body": "{\n\"commandName\": \"IssueComment\",\n\"repositoryFullName\": \"Ultimaker/Cura\",\n\"issueId\": 4206,\n\"IssueComment\": \"{{IssueComment.value}}\"\n}"
> }
> ]
> },
> {
> "name": "Close issue",
> "@type": "HttpPOST",
> "target": "https://api.github.com",
> "body": "{\n\"commandName\": \"IssueClose\",\n\"repositoryFullName\": \"Ultimaker/Cura\",\n\"issueId\": 4206\n}"
> },
> {
> "targets": [
> {
> "os": "default",
> "uri": "https://github.com/Ultimaker/Cura/issues/4206#issuecomment-411868819"
> }
> ],
> "@type": "OpenUri",
> "name": "View on GitHub"
> },
> {
> "name": "Unsubscribe",
> "@type": "HttpPOST",
> "target": "https://api.github.com",
> "body": "{\n\"commandName\": \"MuteNotification\",\n\"threadId\": 365322559\n}"
> }
> ],
> "themeColor": "26292E"
> }
> ]

with Cura 3.2 it is working perfect. I deinstalled Cura3.4.1 an installed it again with the same result.

ChrisTerBeke, 2018-08-09:
That's strange. I'm not aware of any changes in the save to disk feature between those versions.

ianpaschal, 2018-08-10:
Mmm I think we probably did, the problem is more that no one in our office prints by SD card and the bug doesn't reproduce for us otherwise.

These kind of things are usually heavily impacted by the exact computer/hardware being used and very difficult to recreate so it will take time. Also, let me assure you that, "How about an answer" doesn't really make anyone want to work on the issue. 😉

nosailor, 2018-08-10:
Sorry, I didn´t t want to be not polite but English is not my native language. Fact is now:
I have installed Cura 3.4.1 on a second platform and it is working correctly. So the reason for this issue must be my computer. As I´m about to change the computer hardware anyhow You might close this case. Sorry again and thank You.

Ghostkeeper, 2018-10-10:
I know that if we experience an error during the file writing, then we still tend to have an empty file in place. We catch the error but don't remove the file sometimes, it seems. To debug this, we'd need a copy of your Cura log.

nallath, 2019-01-07:
Closing due to inactivity as we never got the requested logs.

closed 2019-01-07 by nallath