Ultimaker/Cura #4208: PLA and PVA not compatible

Tmos88, 2018-08-08:
**Application Version**
3.4.1.

**Platform**
Win 7 64-bit

**Printer**
Ultimaker 3 Extended

**Steps to Reproduce**
Choosing print and support material

**Actual Results**
Cannot choose Ultimaker Black PLA and Ultimaker Natural PVA together, or normal PLA and PVA

**Expected results**
Should work according to Ultimaker

**Additional Information**
PLA + PVA gives compatibility issue on Ultimaker materials. Ultimaker states that these materials should be compatible.

Same compatible issues with normal PLA and PVA

ianpaschal, 2018-08-08:
This is very strange. Indeed they are compatible and it's what we do 99% of our printing with at the Ultimaker office. Can you please share a project file that has this incompatibility in it?

Tmos88, 2018-08-09:
My colleague has the same problem with his own projects. I also tried with different .stl files, and have the same issue. It doesn't seem to be model related.

[UM3E_Part1.curaproject.3mf.curaproject.3mf.curaproject.3mf.zip](https://github.com/Ultimaker/Cura/files/2272823/UM3E_Part1.curaproject.3mf.curaproject.3mf.curaproject.3mf.zip)

ianpaschal, 2018-08-09:
Hmm indeed. That is ultra weird. I think I never noticed it because I tend to just select "PLA" and "PVA" not "Ultimaker Black PLA" and "Ultimaker Natural PVA".

This is indeed a bug. Thanks for reporting.

Tmos88, 2018-08-09:
Same issue with just "PLA" and "PVA".

Tmos88, 2018-08-09:
> Lähetetty: 9. elokuuta 2018 15:22
> Vastaanottaja: Ultimaker/Cura <Cura@noreply.github.com>
> Kopio: Tuomas Suoranta <tuomas.suoranta@valmet.com>; Author <author@noreply.github.com>
>
> Aha yeaaaaaah I just saw that too. And then I thought, "Am I going crazy? I've been printing with PLA + PVA the whole time."
>
> Got a sanity check from a colleague, and it's not a bug. It's because you need to print PVA with a BB core instead of AA. Change your extruder 2 core to BB and all will be well.
>
>
> This email may contain confidential and/or legally privileged information. If you have received it in error, please notify the sender immediately and delete it (together with any attachments) from your system without using or disclosing its contents for any purposes or to any other person. Many thanks for your co-operation.

Great, thanks!

br,
Tuomas