Ultimaker/Cura #4209: Cura 3.4.1 Crashes When Hit Up Arrow in Layer View, cant scroll through layers either

Bulldogger, 2018-08-08:
Says it all
I'm not able to find the App Data log on my Windows 10 machine using the guidance elsewhere on this site.  Would appreciate a steer to it.

fieldOfView, 2018-08-08:
> Says it all

No, it doesn't. Please fill in the issue template.

As for the log, see Help -> Show configuration folder. The file you are looking for is Cura.log.

Bulldogger, 2018-08-15:
Thank you fieldOfView, but I cannot find anything marked .log in the configuration folder.  I did not customize my install in any way, so don't understand why nothing seems to be where it's supposed to be.

Is there another place the log might be?

fieldOfView, 2018-08-15:
What files are in your Configuration folder?

Bulldogger, 2018-10-31:
I've been trying to find this log.  It definitely doesn't show up in that location.  I have no idea why.  I downloaded and installed 3.5.1 and it is still a little odd.  When I click on Layer View sometimes the layers show (the slider shows up and I can slide through) and sometimes I only see the color codes for the types of lines or whatnot.I give up.Alex-----Original Message-----
From: Ghostkeeper 
Sent: Oct 15, 2018 5:24 AM
To: Ultimaker/Cura 
Cc: Bulldogger , Author 
Subject: Re: [Ultimaker/Cura] Cura 3.4.1 Crashes When Hit Up Arrow in Layer View, cant scroll through layers either (#4209)

The log file should be at C:\Users\<your username>\AppData\Roaming\cura\<Cura version>\cura.log, as described in the readme.
This issue cannot be reproduced for me. I just don't see it crashing and if I press the arrow keys while the focus is on the layer view slider, it scrolls through the layers fine.

[
{
"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/Ultimaker/Cura/issues/4209#issuecomment-429771077",
"url": "https://github.com/Ultimaker/Cura/issues/4209#issuecomment-429771077",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
},
{
"@type": "MessageCard",
"@context": "http://schema.org/extensions",
"hideOriginalBody": "false",
"originator": "AF6C5A86-E920-430C-9C59-A73278B5EFEB",
"title": "Re: [Ultimaker/Cura] Cura 3.4.1 Crashes When Hit Up Arrow in Layer View, cant scroll through layers either (#4209)",
"sections": [
{
"text": "",
"activityTitle": "**Ghostkeeper**",
"activityImage": "https://assets-cdn.github.com/images/email/message_cards/avatar.png",
"activitySubtitle": "@Ghostkeeper",
"facts": [

]
}
],
{
"name": "Add a comment",
"@type": "ActionCard",
"inputs": [
{
"isMultiLine": true,
"@type": "TextInput",
"id": "IssueComment",
"isRequired": false
}
],
"actions": [
{
"name": "Comment",
"@type": "HttpPOST",
"target": "https://api.github.com",
"body": "{\n\"commandName\": \"IssueComment\",\n\"repositoryFullName\": \"Ultimaker/Cura\",\n\"issueId\": 4209,\n\"IssueComment\": \"{{IssueComment.value}}\"\n}"
}
]
},
{
"name": "Close issue",
"@type": "HttpPOST",
"target": "https://api.github.com",
"body": "{\n\"commandName\": \"IssueClose\",\n\"repositoryFullName\": \"Ultimaker/Cura\",\n\"issueId\": 4209\n}"
},
{
"targets": [
{
"os": "default",
"uri": "https://github.com/Ultimaker/Cura/issues/4209#issuecomment-429771077"
}
],
"@type": "OpenUri",
"name": "View on GitHub"
},
{
"name": "Unsubscribe",
"@type": "HttpPOST",
"target": "https://api.github.com",
"body": "{\n\"commandName\": \"MuteNotification\",\n\"threadId\": 365969995\n}"
}
],
"themeColor": "26292E"
}
]

Bulldogger, 2018-11-14:
The recently available newer software version seems to have fixed this crashing, whether on purpose or by simple fact of a reinstall.  I did notice that there is sometimes a lag while the layer view is generated, and maybe I was not waiting long enough before hitting keys, but we're talking half a minute sometimes, which does seem long either way.I never could find the Log folder...Alex -----Original Message-----
From: Ghostkeeper 
Sent: Oct 15, 2018 4:24 AM
To: Ultimaker/Cura 
Cc: Bulldogger , Author 
Subject: Re: [Ultimaker/Cura] Cura 3.4.1 Crashes When Hit Up Arrow in Layer View, cant scroll through layers either (#4209)

The log file should be at C:\Users\<your username>\AppData\Roaming\cura\<Cura version>\cura.log, as described in the readme.
This issue cannot be reproduced for me. I just don't see it crashing and if I press the arrow keys while the focus is on the layer view slider, it scrolls through the layers fine.

[
{
"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/Ultimaker/Cura/issues/4209#issuecomment-429771077",
"url": "https://github.com/Ultimaker/Cura/issues/4209#issuecomment-429771077",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
},
{
"@type": "MessageCard",
"@context": "http://schema.org/extensions",
"hideOriginalBody": "false",
"originator": "AF6C5A86-E920-430C-9C59-A73278B5EFEB",
"title": "Re: [Ultimaker/Cura] Cura 3.4.1 Crashes When Hit Up Arrow in Layer View, cant scroll through layers either (#4209)",
"sections": [
{
"text": "",
"activityTitle": "**Ghostkeeper**",
"activityImage": "https://assets-cdn.github.com/images/email/message_cards/avatar.png",
"activitySubtitle": "@Ghostkeeper",
"facts": [

]
}
],
{
"name": "Add a comment",
"@type": "ActionCard",
"inputs": [
{
"isMultiLine": true,
"@type": "TextInput",
"id": "IssueComment",
"isRequired": false
}
],
"actions": [
{
"name": "Comment",
"@type": "HttpPOST",
"target": "https://api.github.com",
"body": "{\n\"commandName\": \"IssueComment\",\n\"repositoryFullName\": \"Ultimaker/Cura\",\n\"issueId\": 4209,\n\"IssueComment\": \"{{IssueComment.value}}\"\n}"
}
]
},
{
"name": "Close issue",
"@type": "HttpPOST",
"target": "https://api.github.com",
"body": "{\n\"commandName\": \"IssueClose\",\n\"repositoryFullName\": \"Ultimaker/Cura\",\n\"issueId\": 4209\n}"
},
{
"targets": [
{
"os": "default",
"uri": "https://github.com/Ultimaker/Cura/issues/4209#issuecomment-429771077"
}
],
"@type": "OpenUri",
"name": "View on GitHub"
},
{
"name": "Unsubscribe",
"@type": "HttpPOST",
"target": "https://api.github.com",
"body": "{\n\"commandName\": \"MuteNotification\",\n\"threadId\": 365969995\n}"
}
],
"themeColor": "26292E"
}
]

closed 2019-01-22 by Ghostkeeper