Ultimaker/Cura #4214: Arrows Stop Working

kencole57, 2018-08-09:
**Application Version**
3.4.0 and 3.4.1

**Platform**
Windows 10 64 bit

**Printer**
Creality Cr-10

**Steps to Reproduce**
Arrows to rotate the build plate stop working, left, right, up down. I click the model, choose a function different to the last one i used, so move if I had used scale or vice versa, click off the model and the arrow keys work again, for a while.  When they stop again, I just repeat this process. I have not been able to determine any specific action that stops the arrow keys from working.

**Actual Results**
Arrow key that stopped working start working

**Expected results**
Arrow keys should not stop working

**Additional Information**
Nil

ianpaschal, 2018-08-09:
Unfortunately this doesn't reproduce for me.

alekseisasin, 2018-08-15:
@kencole57 , could you please determine the actions which will lead to not working arrows. 
Probably you had this,  after adjusting the value in the scale field the cursor stays there and pressing "Left" key on keyboard moves the cursor in the input field and not the build plate. After reaching the beginning (or end) of the input field the build plate starts the rotation.

diegopradogesto, 2019-09-11:
Closing the issue because we couldn't reproduce and lack of extra information.

closed 2019-09-11 by diegopradogesto