Ultimaker/Cura #4217: [3.4.1] Circles strange infill/gapfill

PatrickAntalus, 2018-08-09:
I can't get Cura to do a complete circle.
[image]
[CFDMP_ring.zip](https://github.com/Ultimaker/Cura/files/2274409/CFDMP_ring.zip)

BagelOrb, 2018-08-10:
You could increase the wall line count to prevent this behaviour in this case.

It is a known limitation of how our infill algorithm works.

closed 2018-08-10 by BagelOrb

PatrickAntalus, 2018-08-10:
Thanks for the feedback. I did try 3 walls, but it's the same. Problem is that this third wall does not fit. If this is really standard behavior of Cura, it is unfortunately not usable. In this way, the model needs so much more travel and time.