Ultimaker/Cura #4218: Wrong Initial Layer Height in Gcode

drchopper, 2018-08-09:
**Application Version**

**Platform**

**Printer**

**Steps to Reproduce**

**Actual Results**

**Expected results**

**Additional Information**

ChrisTerBeke, 2018-08-09:
Can you share a project file (.curaproject.3mf) that contains all the settings? Otherwise we can't debug this.

drchopper, 2018-08-10:
[bug.zip](https://github.com/Ultimaker/Cura/files/2278244/bug.zip)

Ghostkeeper, 2018-10-11:
This seems to be fixed in 3.5. I don't reproduce the issue. Here's a few lines of g-code where the initial layer's Z coordinate is set:

    ;LAYER_COUNT:99
    ;LAYER:0
    M107
    G0 F3600 X89.623 Y89.349 Z0.3

Can you also try this in 3.5, Drchopper?

diegopradogesto, 2019-09-13:
Indeed, this is fixed now.

closed 2019-09-13 by diegopradogesto