Ultimaker/Cura #4219: Support Material printed where it is not needed

FynDestiny, 2018-08-09:
**Application Version**

3.4.1

**Platform**

Windows 10

**Printer**

CR-10S
[CCR10_Bobbin Spring.curaproject.3mf.zip](https://github.com/Ultimaker/Cura/files/2275701/CCR10_Bobbin.Spring.curaproject.3mf.zip)

**Steps to Reproduce**

Loaded a few models.
Selected Support Material "Touching Buildplate Only"
Slice

**Actual Results**

As depicted in photos...
[image]
[image]
[image]

**Expected results**

Expected results would be for the support material not be printed where it is not supporting anything 

**Additional Information**

BagelOrb, 2018-08-10:
It is trying to support the top cross sections, but it isn't able to because of the clearance distance or other such support settings.

alekseisasin, 2018-08-15:
You can try to adjust the setting "Support Join Distance" to 0.5mm

FynDestiny, 2018-08-15:
I'll give that a try when I get back onto this part.

Ghostkeeper, 2018-10-12:
In the project file, the problem does reproduce. Setting Support Join Distance to 0.5mm fixes it, so it was the joining of support components that are close together that causes this problem.

Shouldn't it remove support that is not above the build plate BEFORE joining the support areas together?

GregValiant, 2024-11-15:
Is this still an issue with current Cura versions (5.8.0 and up).  Can this be closed?

closed 2024-12-09 by github-actions[bot]