Ultimaker/Cura #4220: Cura 3.2.1 crashes when opening file explorer

Skull802, 2018-08-10:
Ultimaker Cura 3.2.1
Windows 10
Printer: Monoprice Select Mini V2 (E3D)

Step(s) to reproduce:
Just open the file explorer that you use to save/open files through Cura.

Result:
Cura closes within a few seconds with no error, no crash report, or anything. Just shuts down.

Note:
This does not seem to be a bug with certain file types.

I have no idea why this happens. Any help would be appreciated.

Skull802, 2018-08-10:
please i really need help

ChrisTerBeke, 2018-08-10:
Can you try a newer version of Cura to see if it has been fixed for you?

Skull802, 2018-08-10:
I will. I thought 3.2.1 was the newest version.

ChrisTerBeke, 2018-08-10:
we release every 8 to 10 weeks... 3.4.1 is the latest.

ChrisTerBeke, 2018-08-10:
Please share Cura's log file so we can see what happened.

Skull802, 2018-08-10:
what is the log file called? I am having trouble finding it.

ChrisTerBeke, 2018-08-10:
There's info in the readme of this repository about how to find it for your operating system.

ChrisTerBeke, 2018-08-10:
The culprit seems to be this line:

```
UM.Qt.QtApplication.__onQmlWarning [301]: file:///C:/Program Files/Ultimaker Cura 3.4/resources/qml/DiscardOrKeepProfileChangesDialog.qml:81: TypeError: Cannot read property of null
```

Not sure why it's `null` though.

ping @Ultimaker/cura

Skull802, 2018-08-10:
is that how you ping them?
or do i ping a specific person?

ChrisTerBeke, 2018-08-10:
You don't need to ping a specific person. Cura core team looks at GitHub several times a week, so it'll get picked up automatically.

Ghostkeeper, 2018-08-13:
Do you happen to have a Windows Explorer extension, such as Adobe Acrobat or Google Drive showing an entry in the left side of Explorer? If so, there's more information here: https://github.com/Ultimaker/Cura/issues/1537 https://github.com/Ultimaker/Cura/issues/2633 https://github.com/Ultimaker/Cura/issues/2644 https://github.com/Ultimaker/Cura/issues/2731

Skull802, 2018-08-13:
I don't have google drive or adobe acrobat, but one of these might be causing it:
[image]

mqudsi, 2018-08-18:
You should download and run shexview.exe, it'll list all shell extensions.

Ghostkeeper, 2018-09-03:
I see two shell extensions there in the sidebar: Adobe Creative Cloud and Dropbox. Previous bug reports said that uninstalling those fixed the problem. Of course that is not something that Cura should enforce. For now I suggest the workaround to just drop the files from Explorer into the Cura window in order to open them.

But if you'd like to help out testing, you could try uninstalling one of those and see if this is indeed the cause and if so, which extension of the two or if both of them cause this.

DeltaPsiF93, 2018-10-05:
I had to turn off several extensions to get
[image]
 Cura 3.5 to work. Here are the extensions that I turned off.

LipuFei, 2018-10-10:
I feel that this can be related to #4495. We have made a bug fix for this. We have a candidate build for 3.5.1 and could you try it? Here's the link: https://we.tl/t-0gqXweB9KG Thanks!

diegopradogesto, 2018-10-25:
Is this fixed in 3.5.1? Can somebody confirm?

mqudsi, 2018-10-25:
@diegopradogesto why don’t you try and share your results?

Ghostkeeper, 2018-10-26:
Since disabling the shell extension fixed it, this is definitely a duplicate of https://github.com/Ultimaker/Cura/issues/1537 . It has also been confirmed that this is *not* yet fixed in 3.5.1.

closed 2018-10-26 by Ghostkeeper

ianpaschal, 2018-10-26:
> @diegopradogesto why don’t you try and share your results?

In many cases these sorts of bugs are related to someone's specific computer configuration. I'm sure @diegopradogesto would be happy to try if the people experiencing it ship him their computer. 😉

morozsm, 2018-11-03:
Cura crashes when opening Menu -> extension manager or Menu -> Marketplace -> open marketplace on 3.6-BETA.
Confirmed on 3.5, 3.5.1 and 3.6-BETA on recent Fedora 29.
Crashes with:
Auto configuration failed
140471448414016:error:0E079065:configuration file routines:DEF_LOAD_BIO:missing equal sign:conf_def.c:362:line 40
QMutex: destroying locked mutex

Ghostkeeper, 2018-11-05:
Please create a separate issue for that so that we can discuss it properly, @morozsm.