Ultimaker/Cura #4221: [3.4.1] Last layer ironing in vase mode

JasiuHimself, 2018-08-10:
**Application Version**
3.4.1

**Platform**
Windows 10

**Printer**
Creality Cr-10s

**Steps to Reproduce**
Slice anything in vase mode and enable ironing the last layer.

**Actual Results**
Step in last layer and filament spike coming out of it.

[image]

**Expected results**
Smooth last layer

**Additional Information**
I'm not sure if that's a bug, so this might be a feature request not an issue report.

BagelOrb, 2018-08-10:
We could end spiraling the same way it begins: with a flat layer. That layer could then gradually reduce material extrusion because the distance to the last spiralised layer is getting smaller.

smartavionics, 2018-08-11:
I would be surprised if it were possible to do something like that without the result looking either rough or with a lip forming (reverse elephants foot?). Personally, I would rather have the spiral come to an abrupt end which I can smarten up with a scalpel than to have a blobby or fattened top edge.

JasiuHimself, 2018-08-17:
As far as I'm concerned S3D decreases the speed of ascending and the flow gradually along the last layer and the results look really good. The "reverse elephant foot" is almost nonexistent.

[image]

smartavionics, 2018-08-17:
To me, it looks like the top layer is wider than the layer below it. Can you check, please? Also, I would be interested to see the gcode that was printed from. Thanks.

JasiuHimself, 2018-08-17:
Here you go. I forgot to turn the retraction on, so if you try to print it, i would leave plastic smudge at the last layer. 

The last layer is only a little bit wider and only along a half of the last circle.

[image]

[lastLayer.gcode.txt](https://github.com/Ultimaker/Cura/files/2296407/lastLayer.gcode.txt)

smartavionics, 2018-08-17:
Thanks for the gcode. I can see what they are doing. They do one extra lap at the same z-height and decrease the flow to 50%. That still means that it's going to squidge out the excess plastic in the last half a lap. I will do some experiments and see if I can come up with something that doesn't fatten the rim. I am not very hopeful but I will give it some time.

smartavionics, 2018-08-17:
I shall first try doing an extra lap with the flow decreasing to 50% at the half way point and then coasting for some further distance.

BagelOrb, 2018-08-17:
> Be more optimistic about the results :)

I think that with fattening the print will be smoother than without
smoothing altogether.

At lower layer heights you might need higher flows than expected because
the back-pressure is causing more slippage in the feeder.
Also the back-pressure is causing the filament to be compressed a bit in
the Bowden tube, which means you need to feed more material to compensate.
Going down to 50% may actually make a lot of sense.

smartavionics, 2018-08-17:
Well, I have to eat my words as the result came out a lot better than I expected. The point the nozzle leaves the print is at the back of this image (against the yellow). No fattening of the last layer.

[image]

This used flow reduction from 100% down to 50% over 50% of the last outline and then coasted for another 10mm. Layer height is 0.2mm.

I shall upload another release and you can try it yourself

JasiuHimself, 2018-08-17:
Wow! Awesome once again!

smartavionics, 2018-08-17:
I've got stuck on a problem which is I can't reliably detect when the last layer is being printed. For some unknown reason, Cura sometimes prints one less layer than expected. I shall try and work around this.

smartavionics, 2018-08-17:
I worked around the issue and am currently uploading a new release. It should be complete in about 30 mins (sorry, no fibre broadband here!). You can find it in the same location as before.

smartavionics, 2018-08-17:
I should say that you cannot see this final tapering half-layer in the Cura layer view because it will be drawn at the same height as the full layer that preceded it. That's because the layer view thinks all layers are flat rather than gently sloping.

JasiuHimself, 2018-08-17:
Great! Thank you very much!

smartavionics, 2018-08-17:
Hi @JasiuHimself , I have just changed the implementation and have copied the CuraEngine executable to Dropbox. Could you please replace the CuraEngine in the recent archive and give it a test. Thanks.

BagelOrb, 2018-08-17:
> For some unknown reason, Cura sometimes prints one less layer than expected

Wait - what?!
That sounds like a very serious problem!

Does this happen on all objects or only in spiralize mode?

Is it also present in the layer view, or only in the actual print?

smartavionics, 2018-08-17:
Hi Tim, at the moment I am using this test in a LayerPlan method to detect the last spiralized layer:
````
    if ((size_t)layer_nr == (storage.spiralize_wall_outlines.capacity() - 1) || storage.spiralize_wall_outlines[layer_nr + 1] == nullptr)
````

storage.spiralize_wall_outlines has the same capacity as the total number of layers but, sometimes, the last layer doesn't have any data and a LayerPlan for that layer is never created. The slicing tolerance setting (inclusive, middle, exclusive) influences the result but I haven't tried investigating why
a layer goes missing.

smartavionics, 2018-08-17:
And, no, the last layer isn't some tiny point that gets shrunk to nothing!

smartavionics, 2018-08-17:
Maybe what's happening is that the total number of layers is set early on based on the overall height of the object and before the layers have actually been created but when the polygons are created using
the slicing tolerance set by the user, a layer gets shaved off the top or bottom?

BagelOrb, 2018-08-17:
Slicing tolerance could indeed make the last layer empty, but that's by
design then. I don't think that should cause any problems.

Are you ever seeing the problem when slicing tolerance is normal (middle)?

BagelOrb, 2018-08-17:
Yeah that might be what's happening.

So in order to detect whether we are at the top of a spiralise vase we need
to check whether the next layer is empty or something.

smartavionics, 2018-08-17:
>  Are you ever seeing the problem when slicing tolerance is normal (middle)?

Yes, because that's what I use by default.

smartavionics, 2018-08-17:
> So in order to detect whether we are at the top of a spiralise vase we need to check whether the next layer is empty or something.

That's what my test I quote above is doing.

JasiuHimself, 2018-08-20:
Ok, so I just have used your new Cura and the result is amazing.
The antenna is here probably because I used 1mm nozzle and 230 degrees with pla.
[image]

But with a simple cut....
[image]

Once again I'm very impressed with your achievements. 🥇 

The only problem with the print is the first-spiralized-layer-elephant-foot.
[image]

smartavionics, 2018-08-20:
OK, I shall do some experiments and see if the first spiral layer can be tapered without producing any holes.

JasiuHimself, 2018-08-20:
More photos. The top looks really great.

[image]

And the bottom.
[image]

smartavionics, 2018-08-20:
I have a new CuraEngine in Dropbox. Please try and use that. If it hangs up like the other one did. Copy the fdmprinter.def.json that's in Dropbox to cura/share/cura/resources/definitions/fdmprinter.def.json in my tree and restart cura.sh.

This engine now tapers the bottom layer. I have done some tests and so far it appears to create watertight joints but I would be grateful if you could do some testing too.

JasiuHimself, 2018-08-21:
@smartavionics even after substituting the file I'm still "Slicing..."

smartavionics, 2018-08-21:
OK, I will make a new release. Should be there within an hour...

JasiuHimself, 2018-08-21:
Ok, so I have sliced the same model with the same settings with the new release.
There is some difference, now there are two wider first layers.
[image]

For comparision from left to right: Cura 3.4.1, Cura mod @smartavionics , S3D

[image]

S3D has lots of other ugly issues, but (I think so by the way it was printed) they do first solid layers without any offset, and then start spiralizing the same way they end the last layer, with maybe 50% extrusion and a fraction of a layer height, and then proceed from that. Unlike Cura, S3D does that without any nozzle jumps, which, in my opinion, is beneficial for the final outcome.

smartavionics, 2018-08-21:
Hmm, can you please attach the cura project for that print. Thanks.

JasiuHimself, 2018-08-21:
Sure. 
[test.curaproject.3mf.txt](https://github.com/Ultimaker/Cura/files/2306347/test.curaproject.3mf.txt)

smartavionics, 2018-08-21:
Unfortunately, my master branch Cura crashes when I try to load that project file.

Can one of @Ghostkeeper , @jackha , @LipuFei , @ianpaschal say what's wrong here please?
````
2018-08-21 14:06:58,185 - INFO - [MainThread] cura.Settings.MachineManager._updateQualityWithMaterial [1216]: Updating quality/quality_changes due to material change
2018-08-21 14:06:58,186 - DEBUG - [MainThread] cura.Settings.MachineManager._updateQualityWithMaterial [1223]: Current quality type = [normal]
2018-08-21 14:06:58,186 - INFO - [MainThread] cura.Settings.MachineManager._updateQualityWithMaterial [1237]: Current available quality type [normal] is available, applying changes.
2018-08-21 14:06:58,188 - DEBUG - [MainThread] cura.Machines.Models.CustomQualityProfilesDropDownMenuModel._update [15]: Updating CustomQualityProfilesDropDownMenuModel.
2018-08-21 14:06:58,199 - DEBUG - [MainThread] cura.Machines.Models.QualityProfilesDropDownMenuModel._update [50]: Updating QualityProfilesDropDownMenuModel.
2018-08-21 14:06:58,205 - DEBUG - [MainThread] cura.Machines.Models.GenericMaterialsModel._update [23]: Updating GenericMaterialsModel.
2018-08-21 14:06:58,206 - DEBUG - [MainThread] cura.Machines.Models.BrandMaterialsModel._update [86]: Updating BrandMaterialsModel.
2018-08-21 14:06:58,229 - DEBUG - [MainThread] cura.Machines.Models.GenericMaterialsModel._update [23]: Updating GenericMaterialsModel.
2018-08-21 14:06:58,230 - DEBUG - [MainThread] cura.Machines.Models.BrandMaterialsModel._update [86]: Updating BrandMaterialsModel.
2018-08-21 14:06:58,261 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [65]: An uncaught error has occurred!
2018-08-21 14:06:58,262 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [68]: Traceback (most recent call last):
2018-08-21 14:06:58,262 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [68]:   File "/opt/cura/lib/python3/dist-packages/cura/CuraApplication.py", line 871, in event
2018-08-21 14:06:58,262 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [68]:     return super().event(event)
2018-08-21 14:06:58,262 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [68]:   File "/opt/cura/lib/python3/dist-packages/UM/Qt/QtApplication.py", line 411, in event
2018-08-21 14:06:58,263 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [68]:     event._function_event.call()
2018-08-21 14:06:58,263 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [68]:   File "/opt/cura/lib/python3/dist-packages/UM/Event.py", line 189, in call
2018-08-21 14:06:58,263 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [68]:     self._function(*self._args, **self._kwargs)
2018-08-21 14:06:58,263 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [68]:   File "/opt/cura/lib/python3/dist-packages/cura/Utils/Threading.py", line 27, in _handle_call
2018-08-21 14:06:58,263 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [68]:     ico.result = func(*args, **kwargs)
2018-08-21 14:06:58,264 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [68]:   File "/opt/cura/lib/cura/plugins/3MFReader/ThreeMFWorkspaceReader.py", line 582, in read
2018-08-21 14:06:58,264 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [68]:     return self._read(file_name)
2018-08-21 14:06:58,264 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [68]:   File "/opt/cura/lib/cura/plugins/3MFReader/ThreeMFWorkspaceReader.py", line 706, in _read
2018-08-21 14:06:58,264 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [68]:     self._applyChangesToMachine(global_stack, extruder_stack_dict)
2018-08-21 14:06:58,264 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [68]:   File "/opt/cura/lib/cura/plugins/3MFReader/ThreeMFWorkspaceReader.py", line 953, in _applyChangesToMachine
2018-08-21 14:06:58,265 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [68]:     self._applyMaterials(global_stack, extruder_stack_dict)
2018-08-21 14:06:58,265 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [68]:   File "/opt/cura/lib/cura/plugins/3MFReader/ThreeMFWorkspaceReader.py", line 934, in _applyMaterials
2018-08-21 14:06:58,265 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [68]:     machine_material_diameter = extruder_stack.materialDiameter
2018-08-21 14:06:58,265 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [68]:   File "/opt/cura/lib/python3/dist-packages/cura/Settings/ExtruderStack.py", line 77, in materialDiameter
2018-08-21 14:06:58,265 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [68]:     return self.getProperty("material_diameter", "value", context = context)
2018-08-21 14:06:58,266 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [68]:   File "/opt/cura/lib/python3/dist-packages/cura/Settings/ExtruderStack.py", line 103, in getProperty
2018-08-21 14:06:58,266 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [68]:     raise Exceptions.NoGlobalStackError("Extruder {id} is missing the next stack!".format(id = self.id))
2018-08-21 14:06:58,266 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [68]: cura.Settings.Exceptions.NoGlobalStackError: Extruder creality_cr10_extruder_0 #2 is missing the next stack!
````

LipuFei, 2018-08-21:
@smartavionics I'm on it. It's a bug.

smartavionics, 2018-08-21:
Thanks, I appreciate the quick response, good luck!

LipuFei, 2018-08-21:
@smartavionics Should be fixed by af64d91587c4363ca30a9ed38d0ce03eeae701ed. Try to pull the latest master.

smartavionics, 2018-08-21:
Yay! it works again, thanks @LipuFei .

smartavionics, 2018-08-21:
Hi @JasiuHimself , I have looked at the gcode that is produced with this project and it doesn't look wrong. I am thinking that maybe because you are using such a wide line width and also quite deep layers the extruder flow is very high and so when the first spiral layer starts and the flow is reduced in the g-code, the extruder just keeps flowing at the same rate and you get the filament squidging out.

The tests I have done here (with a different model) look very good, no elephant's foot at all.

[image]

smartavionics, 2018-08-21:
Also, looking at your photos, I think your nozzle is too close to the build plate. That first layer is much too low, you should not have all that filament squirting out on the first layer.

JasiuHimself, 2018-08-21:
> I think your nozzle is too close to the build plate.
Yes, that could be the case, I'll try to adjust that.

But still I've printed some similar models and the problem wasn't as visible.

Also, the third model on the right, for which gcode was generated in S3D, was printed on the same printer with no adjustments without elephant foot.

JasiuHimself, 2018-08-31:
Hi @smartavionics, I have adjusted the nozzle - build plate distance. Since then I have printed many vases trying to make them look as good as possible, but there is always a flaw in the beginning of the 
print - the blobs from making the initial spiralized layers watertight and the difference between "solid bottom layers" and spiralized layers.

[image]

Is there any chance you'll give implementing the beginning of spiralization by gradual flow increments a shot and/or adjusting the mentioned visible differences in layers?

Best regards

ianpaschal, 2018-09-03:
@JasiuHimself Sorry, off topic, but love that blue PLA. What brand is it? Looks like blue glass the way the light filters through it.

JasiuHimself, 2018-09-03:
It's from Polish manufacturer [ http://www.plastspaw.pl/index_en.html](url) they are selling it as WOLFix PLA crystal line.