Ultimaker/Cura #4222: Feature Request:  BRIM should (optionally) observe support blocker

sslupsky, 2018-08-10:
**Application Version**
3.4.1

**Platform**
macOS

**Printer**
Prusa i3 Mk3

**Steps to Reproduce**
Create surface with embossed text (reasonably large text size helps see the problem)
Place text surface on the build plate
Add brim

**Actual Results**
When the brim is generated, it generates a brim on the interior of the embossed text (see the "WiTAP" and "SCANIMETRICS" text in the screen shot below.  The "A" and "P" characters in the "WITAP" text has brim generated around the interior of the characters.  Similarly, you can see the same effect in the "A" and "R" characters of the "SCANIMETRICS" text.  

I used "support blocker" to suppress the generation of support for the embossed text.  However, it does not suppress the brim.  I verified that the brim material is not support material by turning off the brim.

**Expected results**
I guess in reality, brim should not create a brim on interior text like this.  But, it may be hard for the slicer to determine that.  So, a workaround would be to use the "support blocker" to prevent brim from putting material where you do not want it.  I suppose if that is not suitable, then create a new tool called a "brim blocker?

[image]

diegopradogesto, 2018-08-13:
This is probably a bug, very easy to reproduce also with 3DBenchy:
[image]

I wouldn't add another feature or extend the "support blocker". This should work fine when indicating "brim only on outside".

BagelOrb, 2018-08-13:
This is intentional behaviour. The brim is on the outside of the inner part of the P.

ianpaschal, 2018-08-14:
I wouldn't say it's intentional behavior. Predictable, yes, but I don't think any of our users really would consider the D shape to be the "outside".

In theory "outside" should be checking if it's fully enclosed in any other polygon. If so, it's not the outside.

sslupsky, 2018-08-14:
Intentional or not, I recommend an option to disable the generation of that material because it is very difficult to remove afterward.  Turning the brim off entirely to avoid it is undesirable.

BagelOrb, 2018-08-14:
The reason I implemented Brim-on-outside-only this way is that there may be
large parts of the bottom which enclose a part which should stick to the
build plate. Cura cannot assume that at a higher layer the inside part of
the D connects to the rest and is therefore less likely to come loose from
the build plate. If you are printing a long extruded D shape then you
really need the inside of the D to have a brim.

Whether it's clear to the user what a setting does is less important than
whether the setting does the right thing.

Adding the option as a third alternative could be a good idea.

Dim3nsioneer, 2018-10-30:
+1 for the option as suggested by @sslupsky .

nallath, 2019-03-01:
For our reference; CURA-5960

bjbeare, 2019-03-08:
I'm also seeing this. It's a pain to remove the brim on internal embossed text. Another alternative is to allow a blocking object (modify settings for infil of other models) to disable brim for the covered area.

topherbuckley, 2019-04-18:
> 
> 
> I'm also seeing this. It's a pain to remove the brim on internal embossed text. Another alternative is to allow a blocking object (modify settings for infil of other models) to disable brim for the covered area.

@bjbeare , I'm not seeing this "disable brim" option within the "modify settings for infill of other models". The entire "Build Plate Adhesion" section does not appear present. Alternatively, as suggested elsewhere, I had some luck with using a skirt with distance to model set to 0. This is slightly different than a brim at sharp corners, but may be a workaround for some.

Ghostkeeper, 2019-04-18:
> @bjbeare , I'm not seeing this "disable brim" option within the "modify settings for infill of other models". The entire "Build Plate Adhesion" section does not appear present. Where would I find this?

This was just a suggestion from bjbeare and is not implemented (yet). I think we'd rather do this stuff automatically rather than asking the user to place blocks around certain parts of the print.

Orum, 2019-11-11:
Not sure if this should be reported as a separate bug or not, but this inside brim also appears when you put an object inside one another.  For example, here are [two rings](https://files.catbox.moe/6fd35b.png), one of which I put a nut inside (as a separate object).  The one with the nut inside it also gets a brim on the inside, even though "Brim only on outside" is checked.

BagelOrb, 2019-11-11:
Yeah that's a known limitation of the current implementation. It's not straighforward how to deal with a brim which might be cut off partly by the outside object.

skrutt, 2020-05-12:
+1
Having "Brim only on outside of" "model" or "shape" option would in my eyes solve this perfectly. When printing most objects this should give the flexibility to get us printing as we'd like, either with brim on outside of model or outside of all shapes in model. 

Of course, there are always edge cases, if someone has an object with a thin tower enclosed in a model with a large base and text on the bottom, tough luck. but there you would have the same options of cad for custom supports just as today. 

But other cases with a thin tower with a large base + text in two different objects, you could place the tower inside the large object in there was a whole for that, but then again you could also keep it outside or just not print them both in the same print if they cant both fit beside each other. 

but having the option of outside shape/outside model should work for the vast majority of users with brim + text on the bottom.

jackharvest, 2020-06-21:
Still holding out and praying for this option to become available.

WimVDK, 2021-08-05:
Just encountered this issue (brim on the inside of letters" myself. Is a fix available yet or perhaps a "brim blocker" feature?

Ghostkeeper, 2021-08-17:
This has not yet been implemented, no. Neither is there any brim blocker (as requested in https://github.com/Ultimaker/Cura/issues/7057). Both are suffering from the same requirement that we'd need to architecturally change the brim from polygons to polylines.

BagelOrb, 2021-08-29:
> If brims are assigned to the nozzle of a part then you could assign a no-print nozzle to take care of brims for areas where you do not want to have brims. Then you just need to add some lines or dots to areas where you want to block brims. Does this sound like something that could work?

Nope. The brim is always printed with one extruder only at the moment.

Kind regards,
Tim Kuipers
PhD researcher at Delft university of technology
Software engineer at Ultimaker

printingotb, 2022-05-04:
Still present in 5.0 beta:
[image]