Ultimaker/Cura #4224: [3.4.1] Relative E not being obeyed at start of layer with firmware retract enabled

zarthcode, 2018-08-11:
**Application Version**
3.4.1 + "Printer Settings" plugin(to manually enable firmware retract)

**Platform**
Windows

**Printer**
Tevo Black Widow (Marlin)

**Steps to Reproduce**
1. Prepare a model to print, with relative e disabled, save to file.
2. Enable "Relative Extrusion" under "Special Modes"
3. Re-slice and export to file

**Actual Results**

G-code, prepared without relative E, on second layer:
`;TIME_ELAPSED:71.009784
;LAYER:-6
M140 S60
G10
G0 X150.603 Y146.132 Z0.584
M204 S3000
;TYPE:SUPPORT
G11
G1 F900 E194.63133
G1 F2250 X151.15 Y146.838 E0.0499
G1 X151.257 Y146.967 E0.00937
G1 X151.974 Y147.758 E0.05965
G1 X152.636 Y148.396 E0.05137
G1 X153.447 Y149.08 E0.05928`

Exact same model, with relative E enabled, on second layer:
`;TIME_ELAPSED:71.009784
;LAYER:-6
M140 S60
G10
G0 X150.603 Y146.132 Z0.584
M204 S3000
;TYPE:SUPPORT
G11
G1 F900 E194.63133
G1 F2250 X151.15 Y146.838 E194.68123
G1 X151.257 Y146.967 E194.69059
G1 X151.974 Y147.758 E194.75025
G1 X152.636 Y148.396 E194.80162
G1 X153.447 Y149.08 E194.8609
`

**Expected results**
With firmware retraction enabled, I don't expect to see the stationary prime/extrusion at all.  However, disabling firmware retract, and enabling relative extrusion causes the same section to look correct:

`;TIME_ELAPSED:71.343735
;LAYER:-6
M140 S60
G1 F9000 E-5
G0 F3600 X150.603 Y146.132 Z0.584
M204 S3000
;TYPE:SUPPORT
G1 F900 E4.99709
G1 F2250 X151.15 Y146.838 E0.0499
G1 X151.257 Y146.967 E0.00937
G1 X151.974 Y147.758 E0.05965
G1 X152.636 Y148.396 E0.05137
G1 X153.447 Y149.08 E0.05928`

It would appear that firmware retraction isn't properly implemented, since it doesn't disable the "manual" retraction command, and it doesn't add an M207/M208 to allow for configuring retract/recover settings.

BagelOrb, 2018-08-13:
I don't understand. Can you tell me the core of the problem? Which line should look different under which settings?

zarthcode, 2018-08-15:
With relative E enabled, the G1 F900 E194.xxxx should not be absolute - (that command extrudes 194mm of filament into a huge blob) and with Firmware retract enabled, shouldn't be present at all.

zarthcode, 2018-08-20:
I've updated the original post to better highlight what's wrong.

BagelOrb, 2018-08-20:
Please attach a project file which results in the problem. Put the .curaproject.3mf file in a zip and drag it in a comment.

This seems to be the very first E operation. In that case relative and absolute E-positions are the same. I'd like to verify by looking at the whole gcode.

mdealer, 2018-09-23:
I'm having the same problem.

`G11
G1 F19980 E95.46055
G1 F3600 X394.602 Y137.141 E0.01419
G1 X394.672 Y137.071
M566 X5400 Y5400
G10
G1 F780 Z0.8
G0 F24000 X392.065 Y126.519 Z0.8
M204 P1000
M566 X1980 Y1980
;TYPE:WALL-INNER
G1 F780 Z0.3
G11
G1 F19980 E95.49553
G1 F3600 X391.953 Y126.033 E0.00829
G1 X391.866 Y125.659 E0.00639
G1 X391.745 Y125.14 E0.00886
G1 X390.801 Y125.14 E0.0157
G1 X390.801 Y112.511 E0.21002
G1 X390.801 Y112.312 E0.00331
G1 X391 Y112.312 E0.00214
G1 X396.301 Y112.313 E0.08816
G1 X396.301 Y125.14 E0.21331
G1 X392.37 Y125.14 E0.06537
G1 X392.296 Y125.443 E0.00519
G1 X392.254 Y125.612 E0.0029
G1 X392.162 Y125.984 E0.00493
G1 X392.065 Y126.519 E0.00282
M566 X5400 Y5400
G10`

Such lines as `G1 F19980 E95.49553` should not be present in relative E and firmware retraction modes.

[Oil_Rig_Ornament.zip](https://github.com/Ultimaker/Cura/files/2408626/Oil_Rig_Ornament.zip)

Ghostkeeper, 2018-11-21:
This bug seems to have been fixed. I can reproduce it with Cura 3.4.1 using that project file, but not in Cura 3.6.0. If you're still encountering it, shoot us a message.

closed 2018-11-21 by Ghostkeeper