Ultimaker/Cura #4225: Issue with nozzle returning to X/Y home in the middle of print and then returning to print.

cyntrifical, 2018-08-11:
So I recently updated Cura to 3.4.1, and I had whatever was the previous version, which worked fine, but to explain the exact problem, my prints will be printing fine, and places where the nozzle hotend should only swing around the print to another part of the print to get around already printed areas, but instead the nozzle assembly seems to autohome only on the X and Y axis (luckily Z axis stays the same) while the extruder continues to pump filament into the hotend, which is causing stringing and lines in my prints. I've gone through all the settings in my printer, and checked my settings in Cura (nothing's changed, I use the same profiles and printer and have not changed a thing since last version.)  Listing info below in template:

**Application Version**
Cura 3.4.1

**Platform**
Windows 10 Home 64bit

**Printer**
Creality CR-10 Original (V1 Motherboard)

**Steps to reproduce**
Not entirely sure, updating to Cura 3.4.1 with the original Creality CR-10 printer???

**Actual Results**
Stringing and lines in print

**Expected results**
Before updating never had too many issues between my software (I run my CR-10 through USB and Cura rather than SD Card and always have) and my printer.

**Additional Information**
This happens randomly so I haven't yet had a chance to make a video or photo of it happening, but as soon as I  can I'll post one.

smartavionics, 2018-08-11:
Hi. Please zip up and attach the cura project that shows the bad extrusion moves. Without that, it's very difficult to guess what could be happening. Thanks.

BagelOrb, 2018-08-13:
Be sure to eject your USB stick before removing it from ur PC.

Ghostkeeper, 2018-10-11:
We've actually seen these lines too. The issue should be fixed in version 3.5. Could you try again with the new version, please?

closed 2020-06-20 by no-response[bot]