Ultimaker/Cura #4226: Deformaties or nickes in layers

FynDestiny, 2018-08-11:
**Application Version**

3.4.1

**Platform**

Windows 10

**Printer**

CR-10S
[CCR10_Bobbin Cog.SLDPRT.curaproject.3mf.zip](https://github.com/Ultimaker/Cura/files/2280456/CCR10_Bobbin.Cog.SLDPRT.curaproject.3mf.zip)

**Steps to Reproduce**

Loaded models
Z axis = 2
One at a Time
Slice

**Actual Results**

You can see in these two photos where I have highlighted, areas where the layers have nicks (bits missing) or bits added on

[image]
[image]

**Expected results**

Expected results are kinda obvious to be honest.

**Additional Information**

ChrisTerBeke, 2018-08-11:
Does this also happen on the actual prints or is it just a visualization problem?

FynDestiny, 2018-08-11:
This particular model is only 10x8.2 mm, it's too small to tell when printed without a way of magnifying it, and I don't have that tech.
It's possible it's just the display rendering, but I have no way of checking the gcode to be sure.
When I load gcode into Cura, I can't select a model to zoom into and examine, let alone view it layer by layer. I don't know if this is a design flaw in Cura or not, but it's bloody annoying.

ChrisTerBeke, 2018-08-11:
I think it's actually showing (and printing) a thinner line in some places to prevent over-extrusion in corners. If the print quality is still OK, I'd say this is expected.

@Ghostkeeper could confirm.

FynDestiny, 2018-08-11:
I await confirmation.

FynDestiny, 2018-08-11:
Sorry to add this, is there a General Question area on this site please?
I want to know if it is possible to get Cura to make, for example, speed or temperature changes within a single print

smartavionics, 2018-08-12:
With regard to the highlighted areas in your photos, most of the glitches you see are problems in the rendering of the layer view. It is, after all, just an approximation of what the printer will be printing. 

However, there is something you should consider which is that the whole printing process (assuming you are using FDM technology) is also an approximation due to the very laggy and imprecise nature of FDM extrusion and the mechanical limitations of the printer. So even if those glitches really are present in the gcode sent to the printer it is unlikely that the visual effect will be the same as what you see on the screen. Very small gaps or over-extrusions will tend to get averaged out.

Even with the improved layer view, Cura is not good at gcode viewing. I always look at gcode using other programs. CraftWare has a great gcode viewer which is still buggy but is a lot better than Cura.

FynDestiny, 2018-08-12:
Fair enough

closed 2018-08-12 by FynDestiny