Ultimaker/Cura #4227: [3.4.1] Cannot connect to 3rd-party printer

Millervill, 2018-08-12:
Hi,
I am using a Da Vinci Jr 1.0 Pro printer. I have recently tried to use it with Cura, however after adding the printer, Cura does not detect it. I know it is not a USB issue nor a computer detection issue as the printer connects to other software and is visible in the devices section on the computer. Im not sure why the software cannot detect my printer. 

Hopefully you can help. 

[image]

ChrisTerBeke, 2018-08-12:
Da Vinci runs proprietary firmware that Cura cannot talk with. You have to use the desktop software provided by them. Unfortunately there's nothing we can do about this.

closed 2018-08-12 by ChrisTerBeke

mishu28nmv, 2019-01-08:
*Only support Cura and Slic3r software (Flavor Selected: RepRap (Marlin/Sprinter))
This is from xyzPrinting site. Tried everything but it doesn't connect!