Ultimaker/Cura #4229: [master] Failed slice leaves Prepare button disabled

smartavionics, 2018-08-12:
**Application Version**

master branch

**Platform**

Linux

**Printer**

Any

**Steps to Reproduce**

Load model then scale up sufficiently so that Cura will refuse to slice it because it's too large. Press prepare button.

**Actual Results**

Cura whines about the model being too big but leaves the Prepare button disabled.

**Expected results**

Prepare button should be enabled again after the failed slice attempt.

**Additional Information**

Couldn't find any action that would re-enable the Prepare - only option is to quite Cura and restart.

smartavionics, 2018-08-12:
Hmm, my money is on 5854ae544f7757103e18af25631a525aef898ec4

alekseisasin, 2018-08-13:
@smartavionics , We know about this issue, I have seen even more cases when it happens. We wanted to introduce a new visual effect/change after the "Prepare" button click. It fails because the "status state" is updated from many places in the backend. Probably it will be reverted. 
Thank you.

Vandrasc, 2018-08-13:
We have a ticket for this, @alekseisasin see CURA-5631 and also have a look at the scenario from the linked issue. 

@smartavionics, thank you for reporting.

smartavionics, 2018-08-13:
I like the idea of the button changing state immediately that it is pressed because, especially with printers with lots of nozzles, there is always some delay before you get some feedback and so the temptation is to press the button again (wrong!)

alekseisasin, 2018-08-13:
It will be nice if it works well. For now, I used to remove the "Preparing" button state because I know at least 3 cases when it does not work.

smartavionics, 2018-08-13:
I'll close this as the problem code has been removed from master.

closed 2018-08-13 by smartavionics